realthunder/FreeCAD_assembly3 #388: MACOSX: The preferences menu can't open preferences window

domomib, 2020-12-21:
In mac OS X, freecad version 2020.12.17 when you try open preferences window trough the menu FreeCAD Link Branch -> Préférences, we do not have access to preferences window, instead we have access to Synchronization view, selection.... see snapshit
[image]

domomib, 2020-12-21:
It is present in the previous version too 2020.12.13

realthunder, 2020-12-22:
That's a strange problem. I can't reproduce it. Can you try change language to English maybe? Can you also paste your `About` information here?

RogerWilko85, 2020-12-22:
Hi,
I have the same problem with this version on OSX BigSur

realthunder, 2020-12-22:
So is this a BigSur only problem? Do you have other version of OSX to confirm?

RogerWilko85, 2020-12-22:
I have on old MBP with Osx El Capitan, and it doesn't run at all. The icon jump one time and that's all.

RogerWilko85, 2020-12-22:
> That's a strange problem. I can't reproduce it. Can you try change language to English maybe? Can you also paste your `About` information here?
I can't change the language, because it's in preference menu.
[image]

domomib, 2020-12-23:
The problem is in my OS X High Sierra  version

domomib, 2020-12-23:
I try remove all FreeCad preferences files, uninstall FreeCad, delete all freecad files and reinstall it again.
The problem is still present. (OS X High Sierra)

realthunder, 2020-12-24:
It is indeed related to language setting. The French translation of menu item `Treeview action` is somehow  `Configuration de l'arborescence`. And Qt will treat any menu item starts with `Configuration', 'Setting', 'Option', etc. as preference menu item. The fix will be available in my next release.

closed 2020-12-24 by realthunder

RogerWilko85, 2020-12-24:
@realthunder Thank you so much ^^

VoidVolker, 2022-07-09:
Similar issue when switching to Russian language:

[image]

[image]

[image]

Installed using brew. Platform: m1.

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.0.1
  * ProDarkThemePreferencePack 1.0.0
[/code]

VladimirKraswov, 2023-11-19:
У меня такая же проблема если выбрать русский язык