realthunder/FreeCAD_assembly3 #390: problems with material editor

kervel, 2020-12-24:
Hello,

when using the latest release from assembly3 i can't use the FEM workbench because it is not possible to set youngs modulus for materials. Whatever you set, it reverts back to 0 mpa which is an invalid setting.

On the 0.19pre appimage it works, see screenshot below (youngs modulus is set correctly)
[image]

On the latest assembly3 appimage it reverts back to zero, and there is a warning about unknown unit in the log
[image]

09:59:02  YoungsModulus in material data seems to have no unit or a wrong unit (reset the value): Steel-Generic
09:59:02  

also the chosen material is always "_document_material" after reopening (but no idea if that's related)

Thanks,
Frank

ceremcem, 2021-02-21:
@kervel Is this issue resolved?

kervel, 2021-02-22:
yes i tried with todays daily and i can once again set youngs modulus!
Thanks
Frank

ceremcem, 2021-02-22:
Would you mind closing the issue?

kervel, 2021-02-22:
ok!

closed 2021-02-22 by kervel