realthunder/FreeCAD_assembly3 #391: Incorrect behaviour of surface generation using b-splines

BeresfordRomeo, 2020-12-26:
FAO@realthunder 

When you use shapebinder or the surface workbench to create a surface between any type of sketch geometry (line or curve) and a b-spline the surface created is incorrect (see attached image). This does not happen between lines or arcs and requires the b-spline to be the upper geometry (z positive). 

In the attached image the (geometrically correct) surface on the left is a shapebinder with a line above (sketch attachment offset z+40mm)  a b-spline. The (geometrically incorrect) surface on the right is a b-spline above (sketch attachment offset z+40mm) a line.

[image]

interestingly if you attempt the surface again using the same geometry the result is different to the first attempt, each time generating an upper geometry higher (z positive) than the last.

This is my first attempt at reporting issues so apologies if I have missed this reported elsewhere or left out important information.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.17.23076 +3024 (Git)
Build type: Release
Branch: LinkStage3
Hash: 560eb2c4645cd548b6f718c645c7af70384ff54c
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: English/United Kingdom (en_GB)

realthunder, 2020-12-29:
This is most likely caused by some obscure OCC bug. The observed behavior is that if a BSpline edge is reused when making a new face, OCC seems to somehow accumulate the edge's transformation (each time the edge is reused).

As shown in the following screencast, if we recompute the bspline sketch, it will re-generate the edge, and when it is used for the first time in making the face in the binder (in other word, if we first recompute the bspline and then recompute the binder), everything is fine. But if we subsequently recompute the binder only, the edge in the sketch will be reused, and we will get incorrect result. Each time a recompute on binder is performed, the bspline edge moves up, which indicates that the edge's transformation is being wrongly accumulated somehow.

Current solution in the referenced commit above is to make a copy of edge only bound shape, which ensures that a new edge is used for each newly made face.

[image]

BeresfordRomeo, 2020-12-29:
Thanks @realthunder  for the information. 

I will proceed and do some more testing on the build I have. I am sure that everyone says this but thankyou so much for the time you take develop and to improve FreeCAD for the benefit of everyone. Your efforts are really appreciated and I will do my best to try to help fund future development.

realthunder, 2020-12-29:
Thanks for the appreciation!

closed 2020-12-29 by BeresfordRomeo