realthunder/FreeCAD_assembly3 #392: Sketcher: offset between mouse cursor and edge/vertex/constraint/etc

invader-zimm, 2020-12-28:
With the latest daily build (2020.12.17), almost every time I try to edit a sketch FreeCAD gets to a state where it's almost impossible to select/edit something or to create constraints, because there seems to be an offset between where the mouse cursor is on screen and where FreeCAD thinks it is.

I'm not exactly sure how to reproduce the issue consistently, but what I usually do is:
- create a new sketch
- create a line/poly line/circle etc
- exit the sketch
- in the viewport, pan/orbit the camera a few times, and also zoom in and out a few times (zooming seems to be the root cause)
- enter sketch edit mode
- zoom in and out a couple of times
- usually, at this point the offset issue appears and things cannot be selected in the viewport

Here's a screen recording, note the preselection highlight appearing when the cursor is nowhere near the corresponding element; when the horizontal constraint value dialog pops up, it's because I double-clicked in empty space...
[image]

I'm not sure, but I think this issue didn't exist in the previous daily build (2020.12.13).

realthunder, 2020-12-29:
I can't seem to reproduce this problem. Any way, please try again with the latest release 12.29.

invader-zimm, 2020-12-29:
I can still reproduce this issue in 12.29 but I think I figured out the key element. I was thinking maybe it has something to do with that particular file I was working with, but I can repro from scratch with a new document:
- make sure FreeCAD starts with perspective rendering mode (go to Edit->Preferences...->Display and set "Camera Type" to "Perspective rendering"; restart FreeCAD)
- new document
- in the Part Design workbench, create a new body and a new sketch; enter edit sketch mode
- draw some lines/circles/etc
- zoom in and out a few times (important step)
- note that so far the preselection highlight follows the mouse cursor correctly
- switch the camera mode to orthographic (important step); you may need to zoom in and out again after switching camera mode
- note that the preselection highlight doesn't match the mouse cursor position

It seems that zooming in and out affects the offset amount and switching from perspective to orthographic triggers the incorrect behaviour:

[image]

Rabbit-sk, 2020-12-29:
Hi there! I also had this problem and it can be reproduced in version 2020-12-29 also on my system.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.29.23076 +3052 (Git)
Build type: Release
Branch: LinkStage3
Hash: c88f329ec8c7d97a32f4e0e5334c4d9876e30eb6
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Here is slightly different procedure with similar outcome:

https://user-images.githubusercontent.com/74265320/103312950-d94b5800-4a1e-11eb-90b4-acda099b27f6.mp4

realthunder, 2020-12-30:
Should be fixed in release 12.30

invader-zimm, 2020-12-30:
Yes, it looks fixed. Thank you!!

Rabbit-sk, 2020-12-30:
I can also confirm that this problem is fixed. @invader-zimm, as the author of this issue, you can mark it as closed.

closed 2020-12-31 by invader-zimm

robertkearns, 2023-11-09:
I am having this issue on Mac for release 0.21.1 using the macbook trackpad