realthunder/FreeCAD_assembly3 #397: Feature request: Shortcut to invoke Formula editor when inserting length in a sketch

Rabbit-sk, 2020-12-30:
Dear Realthunder,

I think it would be convenient if pressing the F key would bring up the formula editor when entering the length, instead of moving the mouse over the small FX button and clicking it.

1.)
[image]

2.) Press the F key

3.)
[image]

Regards

ceremcem, 2020-12-30:
How about inserting `=`? This will fire the formula editor.

Rabbit-sk, 2020-12-30:
Excellent! :D I didn't know about that.

Rabbit-sk, 2020-12-30:
Can I set it to F? I searched here:
[image]
and also here:
[image]
but I did not find setting for this shortcut. It is exposed as a setting somewhere? 

I often use blue construction lines set to equal length, but it is not allways convenient solution. Another scenario is that I'm inserting a string, which is referencing a cell in a spreadsheet. Doing this multiple times in a row by pressing Ctr+V would be more convenient because F is near C, whereas = is far away.

Rabbit-sk, 2020-12-30:
In addition to adding a shortcut setting for opening the formula editor, it would be nice if a description with a shortcut was displayed when pointing the cursor to the round Fx button, so that someone could find out that there is a shortcut to it.

realthunder, 2020-12-31:
The '=' is not handled as a Qt shortcut, to prevent shortcut key interfering with legitimate input characters, I guess. I can add tooltip to the Fx button to mention the '=' shortcut.

irmantus, 2020-12-31:
I was thinking that this whole additional "Formula editor" field is just bad UX design (unesery overhead). You should be able to type vars and formulas directly into the field like it is in Fusion360.

Rabbit-sk, 2020-12-31:
I agree with @irmantus, functionality of these two fields should be merged into one field. But then, how to inform an user about possibility to input a formula? For "backward UX compatibility" old Fx button should be there, alongside the posibility to input vars and formulas directly into the "basic" field.

realthunder, 2020-12-31:
Actually, FreeCAD does have in place expression editing. It is disabled by default because of Qt 4 issue I guess. Maybe it can be reverted now since Qt 5 is now the mainstream. You can try that by setting the following parameter to True. Create it if not existed.

[image]

Rabbit-sk, 2020-12-31:
It is not working.

irmantus, 2021-01-01:
It works but you still need to press "="

Rabbit-sk, 2021-01-01:
Yes, I know, I wrote it only briefly, and later I wanted to describe it more. Here is a comparison of the NoSystemBackground parameter set to false and then set to true:

https://user-images.githubusercontent.com/74265320/103437727-63680c00-4c2b-11eb-9b77-e9c0526a4ecb.mp4

As you can see, it is worse than before, so I will keep this parameter set to false. 

As for the bad UX design you mentioned, I don't mind so much. It would be enough for me to be able to call the Formula editor not only by pressing =, but also by pressing F. 

@realthunder wrote:
> The '=' is not handled as a Qt shortcut, to prevent shortcut key interfering with legitimate input characters, I guess.

If it is not possible to create a shortcut setting for this operation, would it be possible to create a second option how to call up the Formula editor? One would be the original = and second one F. 

PS: I know this is not a big problem, but solving it would improve the user experience.

Rabbit-sk, 2021-01-01:
I forgot `About` info:

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.1230.23076 +3055 (Git)
Build type: Release
Branch: LinkStage3
Hash: c7f24fa0eb98232200ab5cae1ac3be8786a22c2c
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2021-01-02:
Wow, it seems great 👍 I'm looking forward to the next release.

Rabbit-sk, 2021-01-12:
Thank you for the improvements in version 2021.0112. Now the formula button is bigger and has a tooltip. Formula editor appears inplace and can be resized and looks a lot better. Its look is same regardless of the value of the NoSystemBackground key, so I suppose that it is intended to look how it looks now:
[image]
In addition to these improvements, for clearer appearance, maybe there could be some background and border around this improved formula editor, maintaining its inplace behavior. I don't know if this proposal makes sense given the way the formula editor works now or not.

Another new thing is, that Fx button disappears. I don't know what was the intended behavior of hiding it, but it seems to be hiding also when it should not be hidden:

https://user-images.githubusercontent.com/74265320/104365743-a9b94700-5518-11eb-97bd-56b42567de96.mp4

Regards

realthunder, 2021-01-13:
The Fx button is supposed to be shown only when the mouse is over the input field. Somehow it is not hidden on first shown. I'll fix that. The change to show on mouse over is because I have increased the icon size, and made it overlay with the input digits to not waste display spaces. When there is an expression bound to the field, another icon will be shown, and it will stay there.

realthunder, 2021-01-14:
BTW, the expression editor setting is available through preference dialog,

[image]

Rabbit-sk, 2021-01-14:
Thanks for the explanation. These expression editor settings are awesome! 

If `Auto hide editor icon` is ticked, then the Fx button is not hidden on first shown of dialog. After mouse over it behaves as expected. If `Auto hide editor icon` is unticked, then the Fx button shows only on mouse over, so the behavior is also reversed.

https://user-images.githubusercontent.com/74265320/104564768-9c50a980-564b-11eb-947e-b5e294f9e71b.mp4

Rabbit-sk, 2021-02-02:
@realthunder :
> The Fx button is supposed to be shown only when the mouse is over the input field. Somehow it is not hidden on first shown.

@Rabbit-sk :
> If `Auto hide editor icon` is ticked, then the Fx button is not hidden on first shown of dialog. After mouse over it behaves as expected. If `Auto hide editor icon` is unticked, then the Fx button shows only on mouse over, so the behavior is also reversed.

You've probably solved more important issues, I just want to remind you of this little thing. This behavior is present also in version 2021.0202.

closed 2021-02-07 by Rabbit-sk