realthunder/FreeCAD_assembly3 #399: Assembly with multiple instances of the same part

brychanrobot, 2021-01-02:
I'm trying to add multiple links to a screw part. I've tried both explicitly creating multiple links and setting the element count on a link. In both cases I'm able to use PlaneCoincident to place one of the links into the assembly, but when I try to place the second one it breaks, saying that I have inconsistent constraints. The report view gives a stack trace, but it doesn't seemt to contain any helpful information. 

```
15:32:31  8.915702 <asm3.main> assembly.py(3681): solver exception when auto recompute
Traceback (most recent call last):
  File "/tmp/.mount_FreeCA5Ve8qQ/usr/Mod/asm3/freecad/asm3/solver.py", line 109, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/tmp/.mount_FreeCA5Ve8qQ/usr/Mod/asm3/freecad/asm3/sys_slvs.py", line 67, in solve
    raise RuntimeError(reason)
RuntimeError: inconsistent constraints

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 530, in _catch
  File "/tmp/.mount_FreeCA5Ve8qQ/usr/Mod/asm3/freecad/asm3/solver.py", line 416, in solve
    return _solve(*args,**kargs)
  File "/tmp/.mount_FreeCA5Ve8qQ/usr/Mod/asm3/freecad/asm3/solver.py", line 386, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/tmp/.mount_FreeCA5Ve8qQ/usr/Mod/asm3/freecad/asm3/solver.py", line 112, in __init__
    raise RuntimeError('Failed to solve {}: {}'.format(
RuntimeError: Failed to solve DO_subassembly#Assembly001 (Drive): inconsistent constraints
```

I think either I'm missing a step or something is broken in the software. I tried making a simpler assembly to attach to this bug, but it seems to be working just fine, so it would seem that there's something special about my more complex assembly. Both of them included nested assemblies, so I don't think it was that. Any ideas? 
[image]

In the picture you can see that one of the bearing subassemblies is placed, but the other constraint fails, citing inconsistent constraints.

The file is too big to attach here, so DM me if you think it would be helpful, and I can give it to you another way.

realthunder, 2021-01-02:
You can send the file to realthunder.dev at gmail. Can you please paste your FreeCAD about information here?

brychanrobot, 2021-01-02:
OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.1230.23076 +3055 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c7f24fa0eb98232200ab5cae1ac3be8786a22c2c
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-01-03:
In order to use constraint multiplication, the second element must be some circular edge instead of a face. I'll show some error message for that case later. However, there is indeed some other problems in constraint multiplication with your assembly files. After I changed to edge and apply the multiplication, the bearing got flipped. And it is difficult to flip it back. There is definitely some problem there. I'll fix it later.

brychanrobot, 2021-01-04:
Ok. I've adjusted all of the constraints in the drive subassembly to use edges instead of faces, and things seem to work as expected. The bearings are even placed correctly. Thanks for the help! I'll avoid using faces for plane coincident in the future.

luzpaz, 2022-07-01:
@realthunder do we need to keep this ticket open?