realthunder/FreeCAD_assembly3 #401: Bug: Solver fails with two colinear constraints when lines are in different planes. Solver fails to rotate parts into correct orientation.

python50, 2021-01-02:
Today I was testing the collinear constraint.
[image]

I found a very simple situation in which the solver fails. I have made a graphic with embedded text to demonstrate the issue below. I have also attached FreeCAD files to reproduce the issue.

[image]

[colinear bug freecad files.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5760333/colinear.bug.freecad.files.zip)

python50, 2021-01-02:
I believe this issue may be related to #403. Originally, I though this was an issue with the capabilities of the solver. But instead it seems this may be a program logic bug related to "sanity checking" of the applied constraint system. If this is the case, I strongly suspect that the solver would correctly solve the example I provided if this error checking logic was temporarily bypassed for testing purpose.

[image]

python50, 2021-01-03:
I just realized that in my first post I was testing with the wrong branch and and an older daily release. I have reproduced the issue with this repo's latest (2020.1230) daily release.

[image]

[image]

python50, 2021-01-03:
Yikes! I dug deeper only to find (what others know already) that this (and #403) may very well be a bug in the interaction between or within the Solvespace solver library and the Asm3 workbench.

Is it possible that the version of slvs.so packaged with this branch's daily release is an older release? I would like to think that there might be an easy solution to this bug like using a more recent release of the solvespace solver.

ceremcem, 2021-01-03:
See this: https://github.com/realthunder/FreeCAD_assembly3/issues/313#issuecomment-678125785

> Most of the times you need to help the solver a bit. Try to move the bigger part in upper direction. It instantly snaps to the smaller, locked part.

https://user-images.githubusercontent.com/6639874/103483241-e5f0f700-4df6-11eb-85d2-d3ca2192ef76.mp4