realthunder/FreeCAD_assembly3 #402: Feature Proposal: When creating a link to selected object, place new part instance in same location/orientation at selected object. Currently, when creating links of linked objects the new part instance is placed at the same location/orientation as the original part pointed to by the link instead of at the location/orientation of the currently selected object.

python50, 2021-01-02:
Hi,

I was testing making "links" of parts. As a completely new user I found one aspect of the "link" function surprising: the new instance of the linked part can (and is) be initially placed at a different location than the part that was selected to be linked. As it turns out, creating links of linked part instances has a different behavior than creating links from (normal unlinked) parts.
[image]

I have created a graphic below demonstrating the current behavior vs the proposed behavior.

[image]

FreeCAD project file used to demonstrate:
[link_reproduce.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5760533/link_reproduce.zip)

realthunder, 2021-01-26:
When creating a link, I'd assume the user want the instance to be in a different placement, no? If you intention is to have a link that sync with the linked object's placement, it is better to use 'Make Sub-Link' action. If your intention is to replace the part with a Link, well there is 'Replace with Link' action as well. And this shall replace the selected object with a Link at the exact same placement.

ceremcem, 2021-01-26:
According to my understanding, the desired behavior is to have the link object's only the *initial placement* same as the source. 

Think a wheel of a car. Say the wheel appears in a random position within the body assembly when we create first link. You'd modify the placement so that it makes sense to the constraints, then create appropriate constraints. 

When you add second wheel, it would make sense to create the new link next to current one, so that we could easily move it a little (only one move and one rotation without causing any distraction) and then create its constraints.

ceremcem, 2021-01-26:
@python50 Please treat the titles as *titles*, they are not supposed to have the contents.

Rabbit-sk, 2021-01-26:
I don't make assemblies. I use mostly the Part Design workbench, but I also use the link function. 

> When creating a link, I'd assume the user want the instance to be in a different placement, no?

> If you intention is to have a link that sync with the linked object's placement, it is better to use 'Make Sub-Link' action.

I agree. But as @ceremcem wrote, I also think that initial position of new link object could be the same as the position of selected link object used to create another link. It will be not position synced as in the case of Sub-Link, but it will make it easier to place new link object by adding to, or subtracting from values in x, y, z position fields of new link object.

realthunder, 2021-01-27:
New option available to control this behavior.

[image]

ceremcem, 2021-01-27:
Well, there is no such new option as of `b10696c` (3 commits ahead of `c93448e`): 

[image]

But the default behavior is as described here. Even better, IMHO. 

("Random color" is a great facility, btw)

ceremcem, 2021-02-21:
It's now also in the context menu: 

[image]

ceremcem, 2021-02-21:
@python50 Is your issue resolved?