realthunder/FreeCAD_assembly3 #403: Bug: Solver fails trival case with multiple PlaneAlignment constaints when assembling a "cube" from parts

python50, 2021-01-02:
Hello,

I was testing the "PlaneAlignment Constraint" on a few assemblies an noticed any time I used more than one or two instances of the constraint the solver would start failing.
[image]

I have created a graphic below showing the three constraints I placed in the example file. The PlaneAlignment constraints are applied to the "otuside" faces of the cube (the embedded graphic my be misleading since the highlighted surfaces are on the "back" side of the sube with respect to the camera) The constraints that I am placing are in fact solvable, but it appears the solver has a great deal of trouble finding solutions when multiple axes are constraint at the same time. In my previous bug report, I reported on the inability of the solver to "figure out" that it needed to rotate an object. In this test case no rotation is required, only movement in a single axis. However, even though only one part needs moved in one axis to satisfy the constraints the solver has declared that "
inconsistant constraints" have been applied.

Manually moving the object into alignment does not change the behavior in this case. I suspect there is an error in the program logic checking the "solvability" of the applied set of constraints. 

[image]

File to reproduce:
[plane_constraint.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5760577/plane_constraint.zip)

python50, 2021-01-03:
I just realized that in my first post I was testing with the wrong branch and and an older daily release. I have reproduced the issue with this repo's latest (2020.1230) daily release.

[image]

[image]

ceremcem, 2021-01-03:
I suspect that using `Part` container causes something weird. I can not find right now, but I remember @realthunder said that they don't play well with Asm3 right now. I suggest using `Assembly` container instead of `Part` container.

ceremcem, 2021-01-03:
Apparently this issue doesn't have anything with `Part` container. Reproduction (you can skip to 00:56): 

https://user-images.githubusercontent.com/6639874/103484829-ca8be900-4e02-11eb-8661-8bf3fcec102d.mp4

Solution: Create `Plane Constraint`s per 2 elements:  

https://user-images.githubusercontent.com/6639874/103484894-46863100-4e03-11eb-8144-93e5bb2ce011.mp4

However, constraint of your model is not *fully* resolved with this approach.

realthunder, 2021-01-10:
This is a tough problem to solve. I'll explain why. First, you can use three plane alignment constraints to fix any two objects, as shown below. It actually causes over constrain, but asm3 can recognize this simple use case, and auto relax the second plane alignment to plane normal parallel, and the third to a point in plane constraint.

[image]

Now, it becomes difficult when you constrain more than two parts using a single plane alignment constraint, because there are multiple ways to formulate the constraints. What asm3 does is, when adding elements for part A, B, C into the same constraint, it expands the constraint into two as AB, AC. Part B and C will be implicitly aligned through A. This is where the problem is. If there is another plane alignment involving part B and C, it is difficult to recognize and relax it. In fact, if the other alignment is also implicit, it is not possible to relax it. The only way is to change the expansion into AB and BC. This means that we'll need a very complex search algorithm to potentially try all possible combinations in order to fully solve this problem.

In short, try not to use multi-element constraints if it may lead to over constrain. Stick to two-element constraint instead.

python50, 2021-01-10:
Thanks! That makes sense.

luzpaz, 2021-02-23:
@python50 can you test the improvement ?