realthunder/FreeCAD_assembly3 #404: File access on MacOS Catalina worse than upstream

berkakinci, 2021-01-02:
I was trying to test out this branch.  I couldn't access any files with it on MacOS Catalina.
This seems related to the upstream problems, but worse, for some reason.  In Upstream FC, I can't browse most directories, but I can open files if I can specify a full pathname.

What I already do every time I download FC: `codesign --force --deep -s - FreeCAD.RT.20201230.app/`

Launching this "resource" directly (from Finder) seems to access files fine: `/FreeCAD.RT.20201230.app/Contents/Resources/bin/FreeCADLink`

Based on my search of the internet, I found another application having the same issue and their solving it here:
https://github.com/darktable-org/darktable/commit/ccf650756bb86cb9e7bbec24500ec3e0c84b92ec
The TL;DR version, shell scripts to launch the application in MacOS are apparently no longer in fashion.

```
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.1230.23076 +3055 (Git)
Build type: Release
Branch: LinkStage3
Hash: c7f24fa0eb98232200ab5cae1ac3be8786a22c2c
Python version: 3.7.9
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

berkakinci, 2021-01-02:
My apologies; I just realized this is posted in the wrong repo when I couldn't find it.  (But it's where I got the release files.)
I also realized later that "launching from Finder" opened a terminal window too; much like some of the workarounds.  I didn't notice it among my multiple terminal windows.

ceremcem, 2021-02-21:
@berkakinci This issue is still valid, right?

berkakinci, 2021-02-21:
Sorry; I haven't tested it with a newer version since I originally reported it.
Unless something has changed since: There is a real problem, and it seems to be different levels of symptoms with different builds, for some reason.

luzpaz, 2022-06-30:
@berkakinci ping

berkakinci, 2022-07-01:
A lot of things have changed since the initial report, but the spirit of "worse than upstream" remains, albeit slightly.  I downloaded the latest versions to check and found the below.

Note: I no longer run the `codesign ...` line.

**RT FC Release:**
- Open Document dialog box shows user home directory contents (/Users/<username>)
- Open Document dialog does _not_ show contents of user subdirectories: e.g. Documents, Downloads, Desktop
- I can _not_ open a file in Open Document dialog even if I know its path and name.  Once the full name is typed, the file shows up in the dialog, but on an attempt to open the error is "...ios_base::clear: unspecified iostream_category error"
- I can open the same file by drag-drop from Finder into Combo View Model tab.
- There is no prompt for app requesting access to folders at any time.
- The app does _not_ show up in System Preferences -> Security and Privacy -> Privacy -> Files and Folders

**Official FC Release**
- Open Document dialog box shows user home directory contents (/Users/<username>)
- Open Document dialog does _not_ show contents of user subdirectories: e.g. Documents, Downloads, Desktop
- I can _not_ open a file in Open Document dialog even if I know its path and name.  Once the full name is typed, the file shows up in the dialog, but on an attempt to open the error is "<class 'OSError'>: Invalid project file: /Users/..." (This used to work!)
- I can open the same file by drag-drop from Finder into Combo View Model tab.
- There is no prompt for app requesting access to folders at any time.
- The app _does_ show up in System Preferences -> Security and Privacy -> Privacy -> Files and Folders

**RT FC Release Version**
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: C/Default (C)
[/code]

**Official FC Release Version**
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: C/Default (C)
[/code]