realthunder/FreeCAD_assembly3 #406: Zoom to fit

Ebm2003NYC, 2021-01-03:
I have some strange behavior with ''Zoom to fit'', I have some parts where the option work, and others where it doesn't, but I can't reproduce that.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.12.29.23076 +3052 (Git)
Build type: Release
Branch: LinkStage3
Hash: c88f329ec8c7d97a32f4e0e5334c4d9876e30eb6
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: French/France (fr_FR)

ceremcem, 2021-02-21:
@Ebm2003NYC How frequent were you facing with that issue and have you ever faced with it after [the first release following above commit](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11)?

luzpaz, 2021-06-24:
@Ebm2003NYC ?

Ebm2003NYC, 2021-06-24:
Hi,

Sorry for my late reply.
Right now, I don't have this strange behavior with FreeCAD anymore.

Thank you,

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.614.24301 +3584 (Git)
Build type: Release
Branch: LinkStage3
Hash: d1e4d1c5abdedda669e54d1a41b5c08077f7e561
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)

luzpaz, 2021-06-25:
Thanks. Closing ticket

closed 2021-06-25 by luzpaz