realthunder/FreeCAD_assembly3 #407: MacOS unexpected document switching still happening

zardozerr, 2021-01-05:
I did see that one of the fixes for the 12.29 build was a fix for unexpected document switching, but it still seems to happen. It's a bit inconsistent, but I can often make it happen with these conditions:

1. have several documents open. If I'm doing an assembly with lot of external parts, a bunch will be open for example.
2. I go into one of the documents. Make a sketch. Draw a shape, then close the sketcher.
3. It will seem to act normally, then switch to whatever is the top document.

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.1230.23076 +3055 (Git)
Build type: Release
Branch: LinkStage3
Hash: c7f24fa0eb98232200ab5cae1ac3be8786a22c2c
Python version: 3.7.9
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

luzpaz, 2021-02-23:
@zardozerr still an issue in the latest release ?

zardozerr, 2021-02-23:
I haven't been able to test this on the latest build extensively yet. I did get the behavior where it switched back to the 'start' screen at random (if it's left open), so I suspect this may be the same bug. This was on the 2.05.2021 build.

luzpaz, 2021-06-26:
@zardozerr  Please retest with asm3 Daily

luzpaz, 2021-07-30:
@@zardozerr ping

luzpaz, 2021-08-04:
@irmantus can you reproduce this ?

irmantus, 2021-08-04:
I don't work in "assembly with lot of external parts, a bunch will be open for example" to test this. I would need a test file. But this issue sound more or less like #517

luzpaz, 2021-08-04:
@zardozerr can you provide a test project ?

rixth, 2023-04-21:
I can replicate this 100% of the time - on both BigSur & Conda release dmgs. Here are all the exact clicks from boot up:

1. Totally clean install, removed all preferences in ~/Library/Preferences
2. Open FreeCad, "Start" screen displayed
3. Click "Create new empty document" toolbar button
4. Switch from "Start" workbench to "Part Design"
5. Click "Create Body" toolbar button
6. Click "Create Sketch" toolbar button
7. Select "XY_Plane", click "OK" on Tasks tab to create the sketch
8. Drew a rectangle around the origin, didn't constrain, clicked "Close" on tasks tab to complete sketch
9. Returned to my new document as expected
10. Double clicked on my sketch in the tree to edit it
11. *BUG OCCURS* Combo view switches to the "Tasks" tab for my sketch, but the document window shows the "Start" window, and the "Start" tab is selected in the document tab bar.
12. I click "Close" on the sketch task tab, combo view reverts back to the tree for my document, but document window remains on "Start".
13. Manually switch active document back to my new document
14. Double click on sketch to edit it
15. *BUG OCCURS AGAIN* same situation (sketch in tasks tab, "start" in document window)

I can replicate this 100% of the time.

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```

---

1. Totally clean install, removed all preferences in ~/Library/Preferences
2. Open FreeCad, "Start" screen displayed
3. Click "Create new empty document" toolbar button
4. Switch from "Start" workbench to "Part Design"
5. Click "Create Body" toolbar button
6. Click "Create Sketch" toolbar button
7. Click "OK" on Tasks tab to create the sketch
8. Drew a rectangle around the origin, didn't constrain, clicked "Close" on tasks tab to complete sketch
9. Returned to my new document as expected
10. Double clicked on my sketch in the tree to edit it
11. *BUG OCCURS* Combo view switches to the "Tasks" tab for my sketch, but the document window shows the "Start" window, and the "Start" tab is selected in the document tab bar.
12. I click "Close" on the sketch task tab, combo view reverts back to the tree for my document, but document window remains on "Start".
13. Manually switch active document back to my new document
14. Double click on sketch to edit it
15. *BUG OCCURS AGAIN* same situation (sketch in tasks tab, "start" in document window)

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3
Locale: C/Default (C)
```

rixth, 2023-04-21:
A [comment over on the upstream ticket](https://github.com/FreeCAD/FreeCAD/issues/8204#issuecomment-1385986040) had an interesting observation that I confirmed. If I undock the combo view, I can no longer replicate the issue.