realthunder/FreeCAD_assembly3 #408: Daily 20201230 crash on start

janoc, 2021-01-06:
I am getting this when trying to run the latest daily AppImage build:

```
$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20201230-glibc2.12-x86_64.AppImage
/tmp/.mount_FreeCAMJ75lI                                                                                                                                                                                          
FreeCAD 0.19, Libs: 2020.1230R23076 +3055 (Git)                                                                                                                                                                   
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3caf0) [0x7f7f0f83caf0]
#1  /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5XcbQpa.so.5(+0x569d3) [0x7f7f067219d3]
#2  0x7f7f0672183d in QXcbNativeInterface::addHandler(QXcbNativeInterfaceHandler*) from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5XcbQpa.so.5+0x1f
#3  /usr/lib64/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so(+0xd179) [0x7f7f065cc179]
#4  /usr/lib64/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so(+0x957a) [0x7f7f065c857a]
#5  0x7f7f0670d764 in QXcbConnection::glIntegration() const from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5XcbQpa.so.5+0x334
#6  0x7f7f0670e528 in QXcbIntegration::createPlatformOpenGLContext(QOpenGLContext*) const from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5XcbQpa.so.5+0x24
#7  0x7f7f100fc630 in QOpenGLContext::create() from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5Gui.so.5+0x2e
#8  0x7f7f100d766b in QGuiApplicationPrivate::init() from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5Gui.so.5+0x4ab
#9  0x7f7f1055dbc5 in QApplicationPrivate::init() from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib/libQt5Widgets.so.5+0x9
#10  0x7f7f119f3f22 in Gui::GUIApplicationNativeEventAware::GUIApplicationNativeEventAware(int&, char**) from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib64/libFreeCADGui.so+0x12
#11  0x7f7f119f2c53 in Gui::GUIApplication::GUIApplication(int&, char**) from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib64/libFreeCADGui.so+0x23
#12  0x7f7f119f36e7 in Gui::GUISingleApplication::GUISingleApplication(int&, char**) from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib64/libFreeCADGui.so+0x27
#13  0x7f7f1196ced4 in Gui::Application::runApplication() from /tmp/.mount_FreeCAMJ75lI/usr/bin/../lib64/libFreeCADGui.so+0xb4
#14  /tmp/.mount_FreeCAMJ75lI/usr/bin/FreeCADLink(+0x446f) [0x557c67e3e46f]
#15  /lib64/libc.so.6(__libc_start_main+0xeb) [0x7f7f0f826b0b]
#16  /tmp/.mount_FreeCAMJ75lI/usr/bin/FreeCADLink(+0x4789) [0x557c67e3e789]
```

The November daily 20201112 works fine, though. 

This is on Mageia 7, glibc-2.29, running the Nvidia 450.57 driver.

realthunder, 2021-01-07:
Can you please try the latest upstream AppImage and see if it crashes?

janoc, 2021-01-07:
Tested the current upstream daily and it is broken too:

```
$ ./FreeCAD_0.19-23578-Linux-Conda_glibc2.12-x86_64.AppImage 
FreeCAD 0.19, Libs: 0.19R23578 (Git)                                                                                                                                                                              
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020                                                                                                                                              
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.                                                                                                                              
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3caf0) [0x7f67d119daf0]
#1  /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5XcbQpa.so.5(+0x56a06) [0x7f67cc7aea06]
#2  0x7f67cc7ae83d in QXcbNativeInterface::addHandler(QXcbNativeInterfaceHandler*) from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5XcbQpa.so.5+0x1f
#3  /usr/lib64/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so(+0xd179) [0x7f67cbace179]
#4  /usr/lib64/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so(+0x957a) [0x7f67cbaca57a]
#5  0x7f67cc79a764 in QXcbConnection::glIntegration() const from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5XcbQpa.so.5+0x334
#6  0x7f67cc79b528 in QXcbIntegration::createPlatformOpenGLContext(QOpenGLContext*) const from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5XcbQpa.so.5+0x24
#7  0x7f67d1a5b630 in QOpenGLContext::create() from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5Gui.so.5+0x2e
#8  0x7f67d1a3666b in QGuiApplicationPrivate::init() from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5Gui.so.5+0x4ab
#9  0x7f67d1ebcbc5 in QApplicationPrivate::init() from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libQt5Widgets.so.5+0x9
#10  0x7f67d39a2162 in Gui::GUIApplicationNativeEventAware::GUIApplicationNativeEventAware(int&, char**) from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libFreeCADGui.so+0x12
#11  0x7f67d39a0a12 in Gui::GUIApplication::GUIApplication(int&, char**) from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libFreeCADGui.so+0x22
#12  0x7f67d39a18ea in Gui::GUISingleApplication::GUISingleApplication(int&, char**) from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libFreeCADGui.so+0x2a
#13  0x7f67d391608c in Gui::Application::runApplication() from /tmp/.mount_FreeCAwBUXQx/usr/bin/../lib/libFreeCADGui.so+0x13c
#14  /tmp/.mount_FreeCAwBUXQx/usr/bin/freecad(+0x37cd) [0x55711753f7cd]
#15  /lib64/libc.so.6(__libc_start_main+0xeb) [0x7f67d1187b0b]
#16  /tmp/.mount_FreeCAwBUXQx/usr/bin/freecad(+0x4559) [0x557117540559]
```

janoc, 2021-01-07:
D'oh, it is a known problem:
https://forum.freecadweb.org/viewtopic.php?f=3&t=30573

The workaround with unsetting QT_PLUGIN_PATH works for your appimage too (just tested). 

Sorry for the noise.

closed 2021-01-07 by janoc