realthunder/FreeCAD_assembly3 #413: Feature proposal: Add a Refine checkbox to dialogs of additive, substractive and transformation tools

Rabbit-sk, 2021-01-11:
Dear Realthunder,

I was thinking of adding the Refine check box to the dialogs of all additive and subtractive features (or where it is applicable). This could speed up the work. 

The expected behavior of this option would be:
1. Set parameters of a feature and check the Refine checkbox
2. Confirm creation of a feature
3. New shape is created and afterwards Refine will be set to true
4. Later it will be possible in Data tab to set Refine to false

Do you think it would be good to add this option to the dialog?

Regards

realthunder, 2021-01-16:
What about the options below? Is that what you are look for?

[image]

Rabbit-sk, 2021-01-16:
😃 yes! Thank you!
That was what I was looking for:
[image]

realthunder, 2021-01-16:
Be aware that there is come inefficiency in my topo naming encoding regarding this refine operation. It works but the file size may increase rapidly with refine. This has already been fixed in one of my other branch, but it's quite a big change. I'll release it later with a different prefix than 'Daily', or maybe promote the current 'Daily' as stable release. Meanwhile, it will be better to only refine in the last few modeling steps.

Rabbit-sk, 2021-01-16:
One more little thing. The text of this setting:
[image]
could be changed to something like this:
`Automatically refine model after sketch or face based operation`
because this setting has effect also if a face is used:

https://user-images.githubusercontent.com/74265320/104810083-cee8d680-57f2-11eb-82b4-87682733f8b4.mp4

Rabbit-sk, 2021-01-16:
> Be aware that there is come inefficiency in my topo naming encoding regarding this refine operation. It works but the file size may increase rapidly with refine. 

> Meanwhile, it will be better to only refine in the last few modeling steps.

Good to know. 

> This has already been fixed in one of my other branch, but it's quite a big change. I'll release it later with a different prefix than 'Daily', or maybe promote the current 'Daily' as stable release. 

Good idea! I think that this branch deserves a stable release. So far from my experience, I also think that version 2021.0112 is suitable for that. I keep some older daily releases in case there is an issue in the latest release, but new user can download only latest one. It will be good to have a stable release available and can test changes in another release.

Rabbit-sk, 2021-01-26:
> 
> 
> One more little thing. The text of this setting:
> [image]
> could be changed to something like this:
> `Automatically refine model after sketch or face based operation`
> because this setting has effect also if a face is used:
> 2021-01-16.11-55-54.mp4

Do you think that renaming this setting matters? If not, I would like to close this issue, because my first question is solved.

realthunder, 2021-01-27:
I just checked code. The `refine model after sketch-based operation` is actually used exclusively by features in PartDesign, not just sketch-based, while the other `refine` option is for features in Part workbench.

closed 2021-01-29 by Rabbit-sk

Rabbit-sk, 2021-02-10:
> Be aware that there is come inefficiency in my topo naming encoding regarding this refine operation. It works but the file size may increase rapidly with refine. This has already been fixed in one [...]

Is this big change already included in latest daily build, or it is still better to only refine in the last few modeling steps?

realthunder, 2021-02-12:
Unfortunately, no. It's still in my other branch. I'll try to release it this month.

Rabbit-sk, 2021-02-12:
Okay, then please mention it in the release notes :) Thank you for improving FreeCAD!

ceremcem, 2021-02-14:
(Related to (or duplicate of): https://github.com/realthunder/FreeCAD_assembly3/issues/235)