realthunder/FreeCAD_assembly3 #415: Camera moves when creating Pad

Rabbit-sk, 2021-01-11:
Dear Realthunder,

Sometimes when I click to the Pad icon, the camera position changes on its own, sometimes it moves so much, that the selected object for padding is lost from view. Here is a reproduction of the first version of the problem:

https://user-images.githubusercontent.com/74265320/104185144-dab05380-5414-11eb-9b0f-21eed70e41cd.mp4

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2020.1230.23076 +3055 (Git)
Build type: Release
Branch: LinkStage3
Hash: c7f24fa0eb98232200ab5cae1ac3be8786a22c2c
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

This problem is also in the official branch, but I reported it here because I only use your Link branch, as there are many improvements in it, that are important for my work.

Regards

Rabbit-sk, 2021-01-26:
I downloaded version 2021.0126 and found out that now object is not getting lost from view and that is good, but camera do not moves only sometimes, but _everytime_ and makes unconvenient rapid change of 3D view by fitting object into the view, including default 10 mm pad:

https://user-images.githubusercontent.com/74265320/105845170-a3c46b00-5fda-11eb-8110-4040da03d41a.mp4

I'm sorry, I should have described it more clearly. What I wanted was _not to move the camera at all_ when clicking a pad iccon, to have the same camera position, orientation and zoom after clicking pad icon, as it was before clicking it. 

When creating a pad, which dimension is a lot smaller than whole sizse of an object, and want to observe its preview with relation to surrounding objects, the behavior introduced by [a82fc4e](https://github.com/realthunder/FreeCAD/commit/a82fc4effe2794cf1f136e2d211558302adfff1b) manny times will lead to zooming and panning, which has not been needed so far, except for the occasional cases I described at the beginning of this issue (it was not easy to reproduce so I used part of a model I was already working on for some time where it was happening on some objects). To addition, the center for rotation of camera set by View Selection, used before padding, is lost.

https://user-images.githubusercontent.com/74265320/105855609-8eeed400-5fe8-11eb-9ed8-fab453abe73c.mp4

Unfortunately the camera behavior introduced by [a82fc4e](https://github.com/realthunder/FreeCAD/commit/a82fc4effe2794cf1f136e2d211558302adfff1b) is worse than it was before. Please roll this change back, or correct the camera's behavior in the way I've better described now:

> What I wanted was _not to move the camera at all_ when clicking a pad iccon, to have the same camera position, orientation and zoom after clicking pad icon, as it was before clicking it. 

 I apologize again. Thank you in advance for the fix.

realthunder, 2021-01-26:
No apologize needed. And that's not the intended behavior. It must be some setting of yours causes this unexpected camera movement, or maybe some addon. Can you please send me your `link.user.cfg` file for checking? If you are on Windows, it should be located at `C:\User\<Your_UserName>\AppData\Roaming\FreeCAD`.

The intended behavior in that commit is to move camera only if the newly created object is completely outside of the current view, which BTW can also be turned off with a internal parameter `Preferences/Mod/Part/AdjustCameraForNewFeature`.

[image]

Rabbit-sk, 2021-01-27:
Here is my config file:
[link_user_cfg.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5880066/link_user_cfg.zip)

Rabbit-sk, 2021-01-28:
`ShowSelectionOnTop` si great feature, I would like to have it always on, but I turned it off because selecting multiple faces from multiple objects (for creating `Sub-ShapeBinder`)was difficult and often not possible.

Rabbit-sk, 2021-02-02:
Now it behaves mostly correctly. There is only one small issue. If `ShowSelectionOnTop` is disabled, and selected face or Sub-Shape-binder for padding is moved outside of the current view before padding, then camera is not moved (after 0:40):

https://user-images.githubusercontent.com/74265320/106622659-1ce03700-6574-11eb-9d86-7f6af94f36c3.mp4

If sketch is selected and moved outside of the current view before padding, then view is adjusted after clicking on Pad icon.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0202.23662 +3181 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32d19fe309e518dfad042529e631c9729f7fe6d6
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

closed 2021-02-07 by Rabbit-sk