realthunder/FreeCAD_assembly3 #417: Feature proposal: Give a name to a new Body or clone based on the name of the actually clicked Body

Rabbit-sk, 2021-01-11:
Dear Realthunder,

The name of a Link of a body is based on the name of the linked Body:

https://user-images.githubusercontent.com/74265320/104193856-d853f680-5420-11eb-8c06-01122c312134.mp4

But the name of a clone of a body is general:

https://user-images.githubusercontent.com/74265320/104194475-aee79a80-5421-11eb-888a-d32181203663.mp4

Also the name of a new Body is general:

https://user-images.githubusercontent.com/74265320/104194937-464ced80-5422-11eb-96c7-68ec50d54d29.mp4

It is possible to make automatic naming of clones like it is in case of Links? 

In case of a new Body, this is my main interest,  the new body could have name based on a name of an actually selected body in the TreeView:
[image]
[image]

Regards

Rabbit-sk, 2021-01-16:
Is this setting for something I described above?
[image]
I've tried it, but I can't figure out what effect this setting has.

realthunder, 2021-01-17:
No, it is not. I search around the code. This seems to be a abandoned setting. I have repurpose it for this body naming option. I didn't add auto naming to clone, as this is considered as a deprecated feature in my branch and is only here for compatibility with upstream. I'll remove it when merging PartDesign changes into upstream.

Rabbit-sk, 2021-01-19:
> I didn't add auto naming to clone, as this is considered as a deprecated feature in my branch and is only here for compatibility with upstream. I'll remove it when merging PartDesign changes into upstream.

What about projects which uses clones? Will they open correctly after clone function remove?

realthunder, 2021-01-19:
Sure, I mean to just remove the toolbar icon to not confuse the new user.

Rabbit-sk, 2021-01-26:
Now in version 2021.0126 a name given to a new body is based on the name of the body selected before new body creation, but only if that body is used as a base feature for a new body. 

In my workflow, I usually create objects without using the body as a base feature. But I want to give them the same name, with the addition of a sequentional number, because they have the same material. Before export I select them in tree view by name and set them the same color.

So my proposal for naming automation of the new body is:
[image]
[image]
[image]
If no body will be selected before new body creation, the name would be Body.
.
Do you think this behavior could be incorporated into FreeCAD? If yes, should it has an option in preferences?

closed 2021-02-02 by Rabbit-sk