realthunder/FreeCAD_assembly3 #418: Minor improvement - Keeping the Workbench list sorted

leoheck, 2021-01-11:
Hey, I was thinking that it might be good to keep the Workbench list sorted.
See here, the Assembly 3 is being added at the end of the list, it is much better to have this list sorted to find things easier.

[image]

realthunder, 2021-02-20:
I just realized that the previous `workbench sort` functionality conflicts with the workbench list customization (available in `Preferences -> Display -> 3D View -> Sort workbench list`, see realthunder/FreeCAD#109). Therefore, I decide to disable workbench list auto sorting by default. You can still turn it on manually with `Preferences -> Display -> 3d view -> Sort workbench list`.

luzpaz, 2022-01-12:
Should we close this ticket ?

closed 2022-01-12 by realthunder