realthunder/FreeCAD_assembly3 #419: [Question] Will files from this branch be compatible with upstream when relevant features are merged?

jsmaniac, 2021-01-12:
Hi! I'd like to try out the shiny new stuff here, but I'm worried about using it in "production" (hobby) and being unable to migrate files to future releases.

What is the compatibility strategy when merging features upstream? I guess somewhere on this scale:
1. Features here are usually forward compatible with the feature once merged version upstream
2. Incompatibilities, but with a smooth upgrade path (e.g. automatic conversion)
3. A manual change is needed (e.g. replace all `in` units with `"` or vice versa in expressions)
4. This is an experimental playground, files created with this branch are not supposed to be migrated to future mainstream releases.

I hope I don't make it sound like any strategy is better, I'm just trying to gauge whether this branch is intended to be used for short-lived projects that are easy to recreate, or whether it is reasonable to use it for long projects which I'd like to continue working on in a couple of years without starting over from scratch.

I've googled around and glanced at most of the [wiki](https://github.com/realthunder/FreeCAD_assembly3/wiki), but didn't find anything on the topic, just a similar question on the FreeCAD forums; sorry if this is already answered elsewhere!

realthunder, 2021-01-12:
> 1. Features here are usually forward compatible with the feature once merged version upstream

This is the strategy I took. Most of the features are expected to be end up in upstream some day, with the exception of expression Python callables. The worst is that invoking callable is disabled permanently. The more likely situation is that it will be disabled by default, but can be turned on by user.

I haven't thought much about auto migrate the `in` keyword yet, since nobody has complained about this. But, yeah, there should be a way for smooth migration. One way would be to retry parsing after replacing `in` with `inch`.

BTW, thanks for the support on Patron!

jsmaniac, 2021-01-12:
> > 1. Features here are usually forward compatible with the feature once merged version upstream
> 
> This is the strategy I took. Most of the features are expected to be end up in upstream some day,

Wow, that's a fair bit of extra work to ensure compatibility within reasonable expectations…

> with the exception of expression Python callables.

By callables, I guess you mean `.common(…)` in `<<Binder>>.Shape.common(<<Binder>>.Shape).Area > 0` ?

> I haven't thought much about auto migrate the `in` keyword yet, since nobody has complained about this.

This was just an example, I'm using metric units :stuck_out_tongue:

> BTW, thanks for the support on Patron!

My pleasure!

closed 2021-01-12 by jsmaniac

ceremcem, 2021-01-13:
Relevant topic: https://forum.freecadweb.org/viewtopic.php?f=8&t=54298