realthunder/FreeCAD_assembly3 #420: Expression auto-complete and result preview shouldn't invoke callables with side-effects

jsmaniac, 2021-01-12:
Python callables which have side-effects can produce, well, undesirable side-effects when evaluated. When trying to display a preview of an expression's result or get auto-completion entries, the evaluation shouldn't invoke callables which have side-effects. I remember Firebug had the same issue in its JavaScript console back in the days :stuck_out_tongue:.

The auto-complete in the Python console solves that by not invoking any callable, e.g. there is no auto-completion offered after `App.ActiveDocument.getObjectsByLabel("foo")`.

I'd argue it would be best to always run expressions in a sandbox (effectful operations are okay if executed on a temporary document, for example one could clone a sketch into an external document, change its external geometry, and extract the result, without affecting the original document, IIUC this is what Lattice2's TopoSeries does to map parametric copies of a geometry over a list of edges), but this would be difficult to do with FreeCAD as Python and the C++ core don't offer mechanisms to limit side-effects.

It's also in principle a problem with Python properties implemented via `__getattr__`, these could cause side-effects but it is generally implied that properties shouldn't cause side-effects when accessed.

As an artificial example, in the screenshots below, changing the formula text from `._app.ActiveDocument.Sketch040.setConstruction("""Edge2""", False)` to `._app.ActiveDocument.Sketch040.setConstruction("""Edge2""", True)` toggles the edge's construction status.

[image]
[image]

realthunder, 2021-01-15:
I have added a commit to allow disable function evaluation when editing expression. To be precise, the expression editor autocomplete works basically the same as Python console autocomplete, which will not evaluate function. It's result preview function in the expression editor that evaluates the expression while editing.

[image]

jsmaniac, 2021-01-31:
Sorry about the late reply, didn't notice the notification.

Just tested this, it works, and the checkbox is a nice touch. Thanks for the fix! :)

closed 2021-01-31 by jsmaniac