realthunder/FreeCAD_assembly3 #422: Assembly Dependent Boolean Operations / Part Design

ceremcem, 2021-01-17:
# Assembly Dependent Boolean Operations / Part Design

> a.k.a "[Modelling using assembly](https://github.com/realthunder/FreeCAD_assembly3/wiki/Modeling-using-Assembly)"

1. You have a stock `Part` (eg. [3006 trimpot](https://user-images.githubusercontent.com/6639874/104837079-63226e80-58c3-11eb-9430-6826ff962d3f.png)).
2. You need to make some arrangements to mount it in a place inside *Parent_part*: 
    1. Mark the appropriate pin holes on the `circuit board` (= Assembly Based Boolean Operation).
    2. Cut a round hole within the `front panel`, 
        1. Either with a *cutting tool* of assembled part predefined within the stock part. (same as "Assembly Based Boolean Operation", skipping)
        2. Or a custom shape designed inside the *Parent_part* constrained on the assembled `Part` (= Assembly Based Part Design).
3. After making those operations, 
    1. Part can be moved, so the dependent operations should follow the changes (= Move Test)
    2. Part itself might be modified, so the results should track those changes. (= Modify Test)

In the following model, I'm going to cut the `circuit_board` with the *yellow part* (the device, I need its pins) and the `front_panel` with the newly created geometry inside Parent_part:

[image]

Here is the origin of the model: [origin.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5825226/origin.zip)

## Assembly Based Boolean Operation

1. *Teleport* the desired geometries into the root of `Parent_part`:

    > Workaround: Currently a Subshape Binder goes into the active *Body*. So you need to create a temporary `Body` to attract the newly created Subshape Binder. 
  
   1. Create a temporary Body inside `Parent_part`. (This will be automatically marked as "Active Body")
   2. Click the desired geometry (`circuit_board`) + click the `Subshape Binder`.
   2. Click the desired geometry (`device`) + click the `Subshape Binder`.
   3. Verify: The subshape binders went inside the temporary Body within the `Parent_part`. 
   4. Cleanup:
       1. Select those subshape binders, move them outside the temporary Body (select, go to root., `T+D`) 
       2. Remove the temporary Body
3. Create a cut by using those objects (Select `circuit_board` and then the `device`, click to Part -> Cut tool.)
4. Verify that the result shape (Cut_A) is looking nice. 

## Assembly Based Part Design

This seems currently broken. 

## Move Test

1. Move your Part (the `trimpot`)
2. Verify that the cut shape follows the changes (after a Recompute). 

## Modify Test

1. Go to Part
2. Change some geometries which relate to the `Operation A` or `Operation B`. 
3. Verify that the cut shape follows the changes (after a Recompute).

# Screencasts

### Assembly Based Boolean Operation

https://user-images.githubusercontent.com/6639874/104842788-16955e00-58d8-11eb-90f3-fca037fe3a5e.mp4

### Assembly Based Boolean Operation, Move Test

https://user-images.githubusercontent.com/6639874/104843503-0a130480-58dc-11eb-9651-49efbf8d0cfb.mp4

### Assembly Based Boolean Operation, Modify Test

https://user-images.githubusercontent.com/6639874/104842903-dda9b900-58d8-11eb-9ed8-83608f97a54c.mp4

[20210117T1531-operation-a-modify-test-is-successful.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5826063/20210117T1531-operation-a-modify-test-is-successful.zip)

### Assembly Based Part Design + Move Test

https://user-images.githubusercontent.com/6639874/104843800-9540ca00-58dd-11eb-942a-d860494538ff.mp4

### Assembly Based Part Design, Modify + Move Test

https://user-images.githubusercontent.com/6639874/104843862-d89b3880-58dd-11eb-9e0e-8c16e7c0fcf7.mp4

# Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26157 (Git)
Build type: Release
Branch: LinkStage3
Hash: d9d8c77520750e26e625782f89a899f9c7188efd
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Questions

1. Is this workflow still officially supported?
2. If yes, is what is the *Holding `Ctrl` button* (in your original tutorial) come into play? I can't understand the necessity for that, what am I missing?
3. Can we (later) do something about the temporary steps above? It's a little bit confusing. 
4. Why can't I create external geometries within the Part Design WB based on Subshape's rectangular face but I could create based on the circular geometry?

ceremcem, 2021-01-17:
Fixed the wrong namings and the incorrect claim as "currently broken".

realthunder, 2021-01-19:
> * Is this workflow still officially supported?

Yes, that's one of the main purpose of SubShapeBinder.

> * If yes, is when is the _Holding `Ctrl` button_ (in your original tutorial) come into play? I can't figure out the necessity for that, what am I missing?

The tree view drag and drop has been improved to auto detect whether it is possible to drag and than drop. Previously, you are required to use CTRL to indicate the intention to drop without dragging out of the original parent. With the current implementation, it auto try any possible actions. If there are multiple possible actions, you are still required to use modifier key to indicate your intention. You can press CTRL for drop without drag out,  or press ALT for replacing the dropping item with the dragging item (or create a link, if drop to a document item). Without any modifier key, the default action is move, i.e. drag out and drop.

> * Can we (later maybe) do something about the above temporary steps? It's a little bit cumbesome to create and delete the temporary bodies.

I have made some changes to SubShapeBinder creation behavior. It will only go to the body if it is currently active. You can sync my repo and try. Please be aware that I've done a recent merge with the upstream. And the upstream sketch has undergone big changes recently, so something may break. Just report any problem you have. I'll fix it.

> * Why can't I create external geometries within the Part Design WB based on Subshape's rectangular face but I could create based on the circular geometry?

I'm not sure. I guess because for circular geometry it is only one single edge. And sketch right now can only do one edge at a time.

ceremcem, 2021-01-28:
I still couldn't understand what CTRL does. When I use CTRL, the action causes a copy of whole copy of the parent assembly container: 

https://user-images.githubusercontent.com/6639874/106177687-de303280-61a9-11eb-8e9c-305f0c4eec46.mp4

(One time it suddenly crashed without a traceback.)

# Version 
```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26818 (Git)
Build type: Release
Branch: LinkStage3
Hash: b10696c2a46b503de86881509ac98e7e16871eba
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2021-01-28:
In SketchUP, CTRL means "Copy the selected item", does it have the same meaning here?

realthunder, 2021-01-29:
Yes CTRL means copy the selection in most cases. And dropping to a document or some container objects belong to those cases. Dropping to other types of objects may mean another thing, because the target item may not be a container and therefore copying makes no sense. Like for example, dropping to a shape binder with CTRL means to clear all existing binding first and then bind to the dropped object, while without CTRL means to add extra binding. I'll add that information to the binder icon tooltip.

BTW, the icon tooltip is a recent feature. Not many features make use of it right now. But binder is already making some use of that.

ceremcem, 2021-01-29:
What I understand is that the CTRL modifier works according to *your* intention. I still didn't get the point of current behavior but it's OK. If I need to copy a subshape binder, then I'll create a link first and then move the link.

Creating a temporary `Body` to attract the newly created subshape binder trick also isn't that bad. Maybe a context menu would fit here, but it's not urgent. 

A dynamic tooltip that informs the current behavior regarding to the modifier key might be a useful UX decision.

closed 2021-01-29 by ceremcem