realthunder/FreeCAD_assembly3 #423: Expression editor visibility block by scroll bar for long expressions (2021.0112)

DevyHeavy, 2021-01-17:
Hi, when entering long expressions into the expression editor the scroll bar gets in the way. See the below video.

https://user-images.githubusercontent.com/28787282/104857434-99251a00-58e6-11eb-8b30-b2be4f27c18b.mp4

Thanks.

OS: Ubuntu 20.04.1 LTS (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0112.23076 +3101 (Git) AppImage
Build type: Release
Branch: (no branch, rebasing LinkMerge)
Hash: 82adb0b22d37a3be1c88c0c3cc64edf598c315a0
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

snelweg, 2021-01-24:
Same problem here. Made me switch back to the 'official'  release. And then I came straight back to yours!

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0112.23076 +3101 (Git) AppImage
Build type: Release
Branch: (no branch, rebasing LinkMerge)
Hash: 82adb0b22d37a3be1c88c0c3cc64edf598c315a0
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

[image]
[image]

DevyHeavy, 2021-01-24:
Looks like this will be resolved in the next release: https://github.com/realthunder/FreeCAD/commit/2a9dad090665e5613ebeef8cbd02c48d32a95fe5

realthunder, 2021-01-25:
New release available.

BTW, you can also manually resize the editor.

[image]

snelweg, 2021-01-25:
Resizing doesn't work here. I can resize the 'normal' editor, but not the expression editor. Happy to use the new release.

(Thank you for all your hard work, you make FC into what it should be. Very happy with your efforts)
[image]

snelweg, 2021-01-25:
Looks funny. Would be nice if the dialogbox would be broader (as in more width).

[image]

realthunder, 2021-01-26:
Release 0126 added resizing capability for platforms besides Linux. If you are using MacOSX, there is still some problem, which will be fixed in the next release. Currently it requires you to manually add an integer parameter `Preferences\Exprssion\EditDialogBGAlpha` using `Tools -> Edit parameters`, and set the value to 16 or greater. OSX behave strangely if the dialog background is too transparent. Windows works fine with a default alpha value of just 1.

Or, if you prefer normal dialogs, you can disable in-place editing completely in `Preferences -> UI -> Expression editor`.

snelweg, 2021-01-27:
> Or, if you prefer normal dialogs, you can disable in-place editing completely in `Preferences -> UI -> Expression editor`.

This fixed it for me, the other solutions are transparent and not really useful and look odd.

But now there are many strange things happening. The measurements are gone, or very tiny. When I switch back there are no formula's in the edit box anymore, just values. 
I will make screenshots tomorrow, not enough time now.

luzpaz, 2021-07-30:
@snelweg what's the status of this?

snelweg, 2021-07-31:
> @snelweg what's the status of this?

Using FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210717-glibc2.12-x86_64.AppImage and I notice nothing strange anymore. I will close this issue.

DevyHeavy, 2021-07-31:
Yes, this has been addressed, closing.

closed 2021-07-31 by DevyHeavy