realthunder/FreeCAD_assembly3 #425: Construction geometry in sketches turned into defining geometry when opening files from main branch

zardozerr, 2021-01-19:
This happens when opening a project made in main branch, (created in 19.23578, latest at this time of writing). As we all know, there were recently some major changes to the sketcher, so this is probably related to that. The sketches that had blue construction lines broke because those lines were turned into defining lines when opened with linkstage 3 from 20210112. This can majorly break a model of course and cause it to not be able to recompute.

OS: macOS Mojave (10.14)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0112.23076 +3101 (Git)
Build type: Release
Branch: (no branch, rebasing LinkMerge)
Hash: 82adb0b22d37a3be1c88c0c3cc64edf598c315a0
Python version: 3.7.9
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

realthunder, 2021-01-25:
New release available. I have merged against recent upstream which should have solved this problem.

sigxcpu76, 2021-01-25:
Nope, it is not fixed. I've just opened an old document and it is open correctly, while Link branch is converting a construction geometry to concrete.

I've attached such file. `Sketch002` in `Pad002` has 4 construction lines in it.

[tomita mosfet support.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5869591/tomita.mosfet.support.zip)

realthunder, 2021-01-26:
Please test the latest release 0126.

sigxcpu76, 2021-01-27:
Tested. It is resolved. Thanks.

zardozerr, 2021-02-20:
This does seem to be resolved, but not all sketcher issues seem to be (I recently posted another one).

closed 2021-02-20 by zardozerr