realthunder/FreeCAD_assembly3 #426: Assertion error on startup with the PersistentToolbars mod

jsmaniac, 2021-01-19:
When the https://github.com/triplus/PersistentToolbars/ mod is installed, this message appears on startup:

```
Traceback (most recent call last):
  File "<string>", line 239, in onStart
  File "<string>", line 228, in onWorkbenchActivated
AssertionError: No active workbench
```

It seems due to https://github.com/triplus/PersistentToolbars/blob/06dcf6d4086d74bdd20d30fb9f62a220954c552b/InitGui.py#L236-L239 where onWorkbenchActivated() raises an exception instead of returning `None`.

sigxcpu76, 2021-01-19:
Did you notice this?

```Persistent toolbars (FreeCAD 0.16)

Note: Starting with FreeCAD 0.17 persistent toolbars is part of the default FreeCAD experience.```

jsmaniac, 2021-01-19:
> Did you notice this?
> 
> ```
> Persistent toolbars (FreeCAD 0.16)
> 
> Note: Starting with FreeCAD 0.17 persistent toolbars is part of the default FreeCAD experience.
> ```

Didn't notice that, thanks :) . Should I close this or should `Gui.activeWorkbench()` still return None instead of an exception?

realthunder, 2021-01-19:
No, that may break newer code that expects an exception. I don't think we need to change anything, as persistent toolbar will check for FreeCAD version and only activate itself if version < 16, which means it is a deprecated workbench. My release uses date for version, and in this case it is 0112. Unfortunately persistent toolbar is doing string comparison, so '0112' is considered smaller than '16'. I'll make the version something like 112 in my next release to avoid similar problem in the future.

jsmaniac, 2021-01-31:
Fair enough :) I'm closing this as it's a valid "wontfix" :)

closed 2021-01-31 by jsmaniac