realthunder/FreeCAD_assembly3 #427: Ground Rendered Black, Shadows not visible

vasto7, 2021-01-20:
[image]

In Shadow display mode the ground is rendered black.  

Thank you.

My system GPU is a AMD Radeon RX 5700, Driver 20.12.1.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0112.23076 +3101 (Git)
Build type: Release
Branch: (no branch, rebasing LinkMerge)
Hash: 82adb0b22d37a3be1c88c0c3cc64edf598c315a0
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: English/United States (en_US)

snelweg, 2021-01-30:
Do you have a light source somewhere?
Including the file would be handy btw.

vasto7, 2021-01-30:
Thank you for your response.  I did not change any of the defaults.  Just selected shadow from drop down.  Since the object is illuminated, it would seem that there would be a light source.  But happy to check that, if you can let me know how.  When I toggle V,9, it does appear to show as source.  Moving the source changes the illumination of the object, but doesn't change the black ground plane.  Screenshots attached.  Also zip file of model attached.
[image]
[image]
[image]

Zip file attached.
[Seatbelt Tester link.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5898373/Seatbelt.Tester.link.zip)

Kh0rn3, 2021-02-27:
This issue seemed to be related to AMD graphics cards, observed it on a Win10 system with a Vega 64 and a Fedora 33 system with a Ryzen 4500U.

As of build 2021.02.27-2 it appears to be fixed.

vasto7, 2021-02-27:
It does appear to be fixed for me now as well.

Thank you for the notification.  And Thanks to RealThunder for the fix.

vasto7, 2021-03-12:
Issued has returned in latest build.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.311.24301 +3373 (Git)
Build type: Release
Branch: LinkStage3
Hash: 91ca94db328bf6126c4a01547b6aa2202d876ebc
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.5.0
Locale: English/United States (en_US)

realthunder, 2021-03-12:
Oh, that's unexpected. Looks like the problem is related to Qt. Qt 5.15 seems to fix your problem. However it has a more annoying button click [problem](#479) when using stylesheet. If you can live with it, I can make a separate Windows build with Qt 5.15 in next release.

vasto7, 2021-03-13:
No rush at all.  I can live with it.  I wasn't using a stylesheet, so didn't notice any issues there.  Thanks for all your hard work.
Have a great day.