realthunder/FreeCAD_assembly3 #428: Feature proposal: Create link object in active part

Rabbit-sk, 2021-01-21:
Dear Realthunder,

when link is created, in treeview it is in the document root. If a document contains part containers, structure tools to organize  objects, then very likely new object created at the document root will be moved by user somewhere else by drag and drop:

https://user-images.githubusercontent.com/74265320/105316857-bea17480-5bc1-11eb-835e-37c5594d1e44.mp4

To save time and improve the UX, link object should be created in the active part container, also as new body does:

https://user-images.githubusercontent.com/74265320/105316975-ed1f4f80-5bc1-11eb-9786-1b7b3743c8c3.mp4

Regards

Rabbit-sk, 2021-02-01:
Dear RealThunder, maybe this issue is lost among many others, so I would like to remind you. Do you think this behavior could be included in FreeCAD? And if not, what would be the reason for that?

realthunder, 2021-02-02:
New option added. Note that this option is OFF by default.

[image]

closed 2021-02-02 by Rabbit-sk