realthunder/FreeCAD_assembly3 #429: Feature proposal: Tree view function to collapse all expanded objects of the active document

Rabbit-sk, 2021-01-21:
Dear Realthunder,

often during the work, the tree view contains many expanded objects relatively quickly. For easier navigation in the list, it is necessary to manually collapse them one by one, which is lengthy and impractical. I would welcome a feature that would collapse all the expanded objects in the active document with a single click.

Regards

realthunder, 2021-01-21:
Please consult the Qt tree view key bindings [here](https://doc.qt.io/qt-5/qtreeview.html#key-bindings).

For efficient hierarchy navigation, you can use the 'U, U' keyboard shortcut to bring up the hierarchy menu. Hover your mouse over any object either in the tree view or 3D view. If no object under mouse, it will show the hierarchy of the current selection. If you get used to the hierarchy menu, together with the 'T, D' object drag and drop shortcut (which activates drag&drop of the current selection without needing to hold down the mouse button) , you probably won't need the tree view.

Rabbit-sk, 2021-01-23:
> Please consult the Qt tree view key bindings here.

In connection with collapsing branches in tree view I found these two key bindings:
[image]
I already know about the function of the Left key. Using Left key or Minus key does not solve this problem, because I need to collapse all branches at once, not one by one.

On the page you posted I found this: [void QTreeView::collapseAll()](https://doc.qt.io/qt-5/qtreeview.html#collapseAll)

Could be this function included to the functionality of tree view  in FreeCAD?

> For efficient hierarchy navigation, you can use the 'U, U' keyboard shortcut to bring up the hierarchy menu.

Interesting, I did not know about this shortcut. I found out that it is a good way for selecting a body in tree view through 3D view:
[image]
This action leads to selecting a body in tree view like this:
[image]
It is much better than selecting body in tree view by clicking on it in 3D view with sync selection turned on:
[image]
As you can see in the image above, this leads to expanding a body. 

And here I came to the problem that could be also solved. Actually the bahavior of the sync selection leads me to write this issue. Maybe after change in the bahavior of the sync selection, the collapse all branches functionality would be lesss needed, becasue there will be only intentionally expanded branches.

As I described it above, the sync selection behavior is leading to expanding a body in tree view. I don't know how for other users, but for my forkflow the most time this is an unwanted extra step, that is needed to by undone by collapsing it. Maybe this sounds like laziness, but it is not. Spending time for one mouse click, one mouse move, scrolling (body history is sometimes longer) and another mouse click, compared to one mouse click and one mouse move, is big if multiplied by working with larger count of objects. I can imagine that this sync selection behavior could be practical, if one want to edit the parameters of the last feature of a body most time after clicking on it i the 3D view. What I usually ned, is to move that body to another part container, change its color, rename it or copy its name, make a link, hide/unhide. For all this steps expanding the body is not needed. 

So I would prefer the sync selection behavior to be the same as U, U, mouse move, mouse click to select a body from list, because one mouse click is less than this procedure. 

> If you get used to the hierarchy menu, together with the 'T, D' object drag and drop shortcut (which activates drag&drop of the current selection without needing to hold down the mouse button) , you probably won't need the tree view.

I tried it and even though these shortcuts are interesting and the shortcut T, D I have been using for some time, I do not know how these could partly or completely replace the need of using the tree view in my case. In my 3D scene I have numerous part containers and hundrets of objects. Hierarchy menu shows only the part of document hierarchy:
[image]
And that is OK, because showing complete tree view by this shortcut would not make sense.

To summarize, I propose the following two changes:

1. Allow collapsing all expanded branches in the tree view at once.
2. Change the behavior of sync selection to select body in tree view without expanding its branch.

Regards

realthunder, 2021-01-23:
I will add the collapse all shortcut. How about just overload the Asterisk key to toggle the current item?

You can disable (toggle actually) the sync selection behavior anytime using `T, 2` shortcut. You can manually sync the selection using the `Go to selection` shortcut `T, G`. Most of the tree view option is available under menu `View -> Tree view options`.

As to why I think hierarchy menu can replace tree view to some extent, here are some more tips,
* `T, D` works on hierarchy menu items as well. You can initiate drag from an item, and/or drop onto one as well.
* Right click any hierarchy menu item gives you the children, and grandchildren, and so on. Item dropping also works on child item.
* `SHIFT + right click` gives you the object context menu.

Rabbit-sk, 2021-01-26:
> You can disable (toggle actually) the sync selection behavior anytime using `T, 2` shortcut.

Yes I know. I'm toggling it by buton that I added to my custom toolbar:
[image]
But if I want to select body in tree view through 3D view, it must be turned on, and that leads to expanding branch of the particular body.

> Most of the tree view option is available under menu `View -> Tree view options`.

I have tried them all...

> T, D works on hierarchy menu items as well.

I found it too.

> Right click any hierarchy menu item gives you the children, and grandchildren, and so on. Item dropping also works on child item.

Interesting, useful in some cases, but I prefer tree view.

> SHIFT + right click gives you the object context menu.

Another new thing for me, however for me this menu does not contain interesting functions.

> You can manually sync the selection using the `Go to selection` shortcut `T, G`.

I tried it, but maybe I'm missing something here, because it seems to be not working. Lets assume that I do not want to use Sync Selection, because it is expanding body branch. So I will use `T, G` shortcut (instead of one click compared to Sync Selection):

https://user-images.githubusercontent.com/74265320/105861187-bba5ea00-5fee-11eb-998f-4284b0a7e36c.mp4

As you can see, nothing happens.

> I will add the collapse all shortcut. How about just overload the Asterisk key to toggle the current item?

That is a good idea. 

> `Gui: support recursive expand/collapse in tree view `

How does it work? Can you provide a screencast showing also key presses? I tried it and it seems that branches are not collapsed when expected:

https://user-images.githubusercontent.com/74265320/105863558-4ab40180-5ff1-11eb-9606-8aa6415bb62e.mp4

I think that using Asterisk key to expand and collapse branches of items under particular item (Document root, Part container, Group, Body, Feature) will be enough. The only exception for expansion may be Origin branches, which have 6 items and occupy a significant number of rows of the tree view. I don't think it would usually be necessary to expand them all at once:
[image]

realthunder, 2021-01-26:
The `Go to selection` command is broken. I've just fixed it.

I didn't overload `Asterisk` in the end, but choose to overload `Left/Right, Minus/Plus`. Pressing on already collapsed/expanded item will recursively collapse/expand it. So for example, double pressing `Minus` will recursively collapse an item. Note that the `Left/Right` key can also navigate between columns, so to be sure, triple pressing the key will definitely recursively collapse/expand an item.

Rabbit-sk, 2021-01-27:
I tried the Left and Right keys again. It works as you described. Could opening an older version before trying a newer one cause strange behavior? Now, after booting the PC, I only opened a new version and it works fine.

Rabbit-sk, 2021-01-27:
What about not expanding (but collapsing yes) of `Origin` branches? I think that they take up a lot of space and that in most cases it will not be necessary to expand them in all bodies under a part container. Usually, when the body is active and the user wants to create a sketch or dateúm plane bound to one of the origin planes, the user expands `Origin` branch.
[image]
vs
[image]
But this is not very important thing, it is just a suggestion.

Rabbit-sk, 2021-01-27:
It can be very useful to mention this new recursive expand/collapse function and the associated shortcuts with overloaded `Left/Right`,` Minus/Plus` keys in the context menu of the tree view items, so that the user can easily find out that this function exists.
[image]

Rabbit-sk, 2021-02-02:
> It can be very useful to mention this new recursive expand/collapse function and the associated shortcuts with overloaded `Left/Right`,` Minus/Plus` keys in the context menu of the tree view items, so that the user can easily find out that this function exists.

Do you think it is appropriate to fill the another line in this context menu with such a function? If not, I would like to close this issue.

realthunder, 2021-02-02:
I think the context menu is already too crowded. I tried with the tooltip and even status bar message for the hint, but they all feel too intrusive and not that useful.

Rabbit-sk, 2021-02-02:
> but they all feel too intrusive and not that useful.

I can imagine. So then it's better to leave it as it is.

closed 2021-02-02 by Rabbit-sk