realthunder/FreeCAD_assembly3 #431: [Question] Is there an official way to eval a FreeCAD Expression from Python (e.g. from a macro)?

jsmaniac, 2021-01-27:
Hi!

I have found this allows evaluation of a Python-like FreeCAD Expression. However it is just a hack, and requires setting up a property with the correct type (the code below only works for length expressions, it won't work for an angle).

**Is there a proper way to do this?** I cound't find anything in this branch's wiki and I'm pretty sure it's not possible in stock FreeCAD.

With the added power of the Python-like Expressions, it becomes interesting to evaluate expressions in other contexts than a simple property (e.g. loop an expression over values of `i`).

```
def evalFreeCADExpression(expr):
  b = App.ActiveDocument.addObject("Part::Box","Box")
  b.setExpression('.Placement.Base.x', expr)
  b.recompute()
  ret = b.Placement.Base.x
  b.Document.removeObject(b.Name)
  return ret

evalFreeCADExpression("2cm*3mm")
```

Thanks :)

realthunder, 2021-01-27:
Currently, an expression must be bound to some document object. You can use that object's API `evalExpression()` to obtain the result. See below for an interesting example. Note that once the bound object is deleted, the expression will cease to work.

```python
expr = App.ActiveDocument.Box.evalExpression('lambda x : _py.print(x)')
expr(123)
```

jsmaniac, 2021-01-28:
Thanks, that's much better than my approach, don't know how I missed the `evalExpression()` !

It makes sense that the evaluator needs to be attached to an object, since expressions allow references to the "local" or "current" object.

luzpaz, 2022-06-30:
Closing ticket

closed 2022-06-30 by luzpaz