realthunder/FreeCAD_assembly3 #432: Roadmap to Collision Detection

ceremcem, 2021-01-27:
Is it possible to project a roadmap to collision detection feature? (Related to #93)

realthunder, 2021-01-28:
How to you plan to use the collision detection function? I don't have any immediate plan for that. I'll probably look for someone for collaboration. This [thread](https://forum.freecadweb.org/viewtopic.php?f=18&t=39165&start=200) looks interesting. I'll ask around.

ceremcem, 2021-01-28:
Actually I personally have nothing urgent. I'm still on "Paid support of Assembly3" and current Solidworks requires that feature (which is already in Solidworks). However, that is also not urgent because there are a lot of different tasks where Assembly3 nicely fits as of this moment. I just wanted to have an idea of this topic. 

I'm going to close this issue as it's not a objective obstacle for any task at this moment, not planned for near future.

closed 2021-01-28 by ceremcem