realthunder/FreeCAD_assembly3 #433: [Feature request] Get the Abstract Syntax Tree of an expression from Python (e.g. a macro or WB)

jsmaniac, 2021-01-27:
Context: I'm going to work on a nodal interface for FreeCAD (PyFlow-like but without the security issues). It will be an editable representation of the document's dependency graph (input pins for object properties and the .Group special property, output pin for the object itself to be connected to .Group or App::PropertyLink). I'd like to show expressions as small nodes which can be expanded to show the expression as a graph, and for that it would help a lot to access the AST :slightly_smiling_face:

realthunder, 2021-01-28:
Are you compiling your own binary? If so, you can try the above mentioned API. Check the python docstring for details. I did it in a hurry, not fully tested it yet, because I need to make a release for some urgent fix.

jsmaniac, 2021-01-28:
That's awesome, just what I needed :)

I was avoiding building my own binaries so far to save myself some sysadmin headaches, might be an excuse for me to attempt a build and maybe script it so that others can get a fresh release using a one-liner :)

Closing this since it's implemented now :p that was fast!

oficinerobotica, 2021-06-08:
@jsmaniac would you consider this issue solved? If so could you please close it? Cheers

closed 2021-06-08 by jsmaniac