realthunder/FreeCAD_assembly3 #435: Spreadsheet: Unable to set alias with non-alphanumberic characters

Kirkov22, 2021-01-29:
I'm actually seeing two issues with separating aliases, but I'm unsure if they are related.

1) Any attempt to set an alias with a non-alphanumeric character ( '_', '-', '.', etc.) result in the following error:
  **<class 'ValueError'>: Invalid alias**
  Up till this version I was able to set alias names with underscores.

2) I cannot select a cell and then click on 'Alias' through the UI if it doesn't already have an alias. It looks like the selection has become limited to whatever text already exists in that space in the UI. I can still right click and select Alias to open up the dialogue though.

These issues are present in any sheet I try to create and edit with this version; existing project or new.

Thank you for all the work you do on this project!

**Version:**

OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0128.23662 +3163 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 46bd4d7710ce2c2e09c2f898eaa1a563327c3911
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

realthunder, 2021-01-29:
Thanks for reporting. The first issue should have been fixed. But note that only middle underscore is allowed as before. Leading underscore is not allowed to avoid conflict with some internal properties.

I can't seem to reproduce the second issue, though. In fact, it behaves as such in upstream, and I've specifically allowed setting alias in empty cell in 01.28. Can you please provide a sample file showing the problem?

realthunder, 2021-01-29:
BTW, 01.28 has some serious problem related sketch external geometry. Please do not use it for your project. I am currently making a new release.

Kirkov22, 2021-01-29:
I feel foolish about the second issue, it looks like I can select the field to edit I just have to be more precise with the mouse pointer. I did see your patreon post about version 01.28 before, happy to wait for next release! Thanks again for being so quick to tackle these issues!

Kirkov22, 2021-01-29:
Verified the issue is gone as of the 01.29 build

closed 2021-01-29 by Kirkov22