realthunder/FreeCAD_assembly3 #437: FreeCAD draft grid shows in other WB's, and it shouldn't.

AnonymousFromEgypt, 2021-01-29:
Hello,
- of course grid and snapping should be a common and separate toolbar in most of WB's just like File and view etc.
but since it is dedicated to Draft WB so it shouldn't appear in other WB's.

Linux mint 20.1
FreeCAD 0.19_pre daily 1.29
Video for the issue
[https://www.youtube.com/watch?v=qIeERaFORuY](url)

realthunder, 2021-01-30:
@yorikvanhavre Hi, Yorik. Could you please take a look at this. I also notice this behavior for quite some time. Is this intentional?

snelweg, 2021-01-30:
I noticed it too, there are even some questions on the forum about this (solution was to use a hot-key to stop the display of the grid ....)

AnonymousFromEgypt, 2021-01-30:
I guess this bug started from th begining of freecad coding accoring to a policy it says:
if a repeated function existed in one WB so there is no need to add it to another to save coding effort and time.
and this is WRONG.
FreeCAD is aweome software but some UI bugs gives bad indication.
Thanks to realthunder and his efforts to correct this impression.

FreeCADAntiBugs, 2021-02-02:
Hello @yorikvanhavre, 
i hope you read what realthunder wrote to you !! 
can you make independent grid & snap toolbar works for all WB just like File, View, Edit .. etc ??
thank you.

ceremcem, 2021-02-22:
@realthunder I'm labeling this issue as `upstream`.