realthunder/FreeCAD_assembly3 #438: Set material does not work

snelweg, 2021-01-30:
When I try to set a material to an object (Arch WB) then I get the following error (and nothing happens)"

> 11:33:40  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAhgqOq2/usr/Mod/Arch/ArchMaterial.py", line 469, in setEdit
    self.taskd = _ArchMaterialTaskPanel(vobj.Object)
  File "/tmp/.mount_FreeCAhgqOq2/usr/Mod/Arch/ArchMaterial.py", line 522, in __init__
    self.form.ButtonSectionColor.setIcon(QtGui.QIcon(colorPix))
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'ButtonSectionColor'

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0129.23662 +3166 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: bd2d90d312b3519573a323f97d7544d82f0e314d
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

snelweg, 2021-02-02:
Looks like it's an error in the main branch, but there is not much enthousiasm to fix it there. I only get the usual bureacratic crap.

Topic:
> https://forum.freecadweb.org/viewtopic.php?t=43989

Solution:
> https://forum.freecadweb.org/viewtopic.php?t=43989#p375139

Would be nice if this was fixed in the superior Linkstage3 branch.

realthunder, 2021-02-03:
These are two problems right? I can't seem to reproduce the first Arch material problem. Can you please provide a file showing the problem? You second post seem to refer to an external module that is under development? I don't think anything can be done on my side.

snelweg, 2021-02-03:
> These are two problems right? 

No, it's the same problem.

>I can't seem to reproduce the first Arch material problem. Can you please provide a file showing the problem? 

It happens with all files, new files, old files. It happens all the time. Looks like something is missing from the Arch WB. I will delete  my .freecad directory and see if it starts working then.

>You second post seem to refer to an external module that is under development? I don't think anything can be done on my side.

It was kinda a solution to the problem, but please ignore that post. It's not really relevant.

snelweg, 2021-02-03:
Deleted the .freecad folder. Started the latest LinkStage3 branch and the same message and problem:

> 11:29:23  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAFNakOc/usr/Mod/Arch/ArchMaterial.py", line 469, in setEdit
    self.taskd = _ArchMaterialTaskPanel(vobj.Object)
  File "/tmp/.mount_FreeCAFNakOc/usr/Mod/Arch/ArchMaterial.py", line 522, in __init__
    self.form.ButtonSectionColor.setIcon(QtGui.QIcon(colorPix))
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'ButtonSectionColor'

>OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0202.23662 +3180 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 60f2599dea856906f4d59cc091e0713f6049fe2a
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
[FC_arch_material.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5917757/FC_arch_material.zip)

ceremcem, 2021-02-03:
@snelweg What are we expecting with this attachment? It opens correctly, doesn't need a recompute and forcing a recompute does not trigger any error. Can you please give instructions to reproduce the problem?

snelweg, 2021-02-03:
> @snelweg What are we expecting with this attachment? It opens correctly, doesn't need a recompute and forcing a recompute does not trigger any error. Can you please give instructions to reproduce the problem?

Aha, sorry I thought it was obvious.

The problem is that when I try to use the materials from the Arch WB the reportview gives me that error message and no material functions can be used.

snelweg, 2021-02-03:
https://user-images.githubusercontent.com/1257901/106741308-ca5a5580-661b-11eb-8009-63a1c3e2c328.mp4

ceremcem, 2021-02-03:
I can't reproduce the error: 

https://user-images.githubusercontent.com/6639874/106754320-e40c9480-663d-11eb-8873-dd0e11ef73a7.mp4

## Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26844 (Git)
Build type: Release
Branch: LinkStage3
Hash: a3a549589c8e6d2e888ac004a7fc534f72852b18
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2021-02-03:
In case you want to build LinkStage3 by yourself: https://github.com/ceremcem/build-freecad-asm3

realthunder, 2021-02-04:
This turned out to be a build cache problem. Note sure why, but it seem only appeared on Linux.

snelweg, 2021-02-04:
> This turned out to be a build cache problem. Note sure why, but it seem only appeared on Linux.

Ok, thank you for pursuing this problem.
You are a gentleman.

snelweg, 2021-02-06:
Using the 2021.0205.23662 +3201 version I can confirm the material dialog box is working again.

>OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

closed 2021-02-06 by snelweg