realthunder/FreeCAD_assembly3 #439: Image updater shows 20210130 version but will not install it

snelweg, 2021-01-30:
The 20210130 release is not visible on the releases github page btw.

> loading the plugin: True
   INFO:   setAppImage :  "/run/user/1000/appimagelauncherfs/0001.AppImage" . 
   INFO:   setAppImage :  "/home//Applications/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210129-glibc2.12-x86_64_39ebd499342219ec96fd253bc3c472af.AppImage" . 
   INFO:   getInfo : AppImage is confirmed to be type 2. 
   INFO:   getInfo : mapping AppImage to memory. 
   INFO:   getInfo : AppImage architecture is x86_64 (64 bits). 
   INFO:   getInfo : updateString( "gh-releases-zsync|realthunder|FreeCAD_assembly3|latest|FreeCAD-asm3-Daily-Conda-Py3-Qt5-*-x86_64.AppImage.zsync" ). 
   INFO:   getInfo : finished. 
   INFO:  setControlFileUrl : using github releases zsync transport. 
   INFO:  setControlFileUrl : github api request( QUrl("https://api.github.com/repos/realthunder/FreeCAD_assembly3/releases/latest") ). 
   INFO:   handleGithubAPIResponse : starting to parse github api response. 
   INFO:   handleGithubAPIResponse : http response code( 200 ). 
   INFO:  handleGithubAPIResponse : latest version is  "0.11" 
   INFO:  handleGithubAPIResponse : asset required is  "FreeCAD-asm3-Daily-Conda-Py3-Qt5-*-x86_64.AppImage.zsync" 
   INFO:  handleGithubAPIResponse : inspecting asset( "FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage" 
   INFO:  handleGithubAPIResponse : inspecting asset( "FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage.zsync" 
   INFO:   setControlFileUrl : using  QUrl("https://github.com/realthunder/FreeCAD_assembly3/releases/download/0.11/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage.zsync")  as zsync control file. 
   INFO:   handleGithubMarkdownParsed : starting to parse github api response. 
   INFO:   handleGithubMarkdownParsed : http response code( 200 ). 
   INFO:   getControlFile : sending get request to  QUrl("https://github.com/realthunder/FreeCAD_assembly3/releases/download/0.11/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage.zsync") . 
   INFO:   handleControlFile : starting to parse zsync control file. 
   INFO:   handleControlFile : http response code( 200 ). 
   INFO:   handleControlFile : searching for checksum blocks offset in the zsync control file. 
   INFO:   handleControlFile : found checksum blocks offset( 306 ) in zsync control file. 
   INFO:   handleControlFile : zsync make version confirmed to be  "0.6.2" . 
   INFO:   handleControlFile : zsync target file name confirmed to be  "FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage" . 
   INFO:   handleControlFile : zsync target file MTime confirmed to be  QDateTime(2021-01-30 06:05:35.000 CET Qt::LocalTime) . 
   INFO:   handleControlFile : zsync target file blocksize confirmed to be  4096  bytes. 
   INFO:   handleControlFile : zysnc target file length confirmed to be  743968808  bytes. 
   INFO:   handleControlFile :  3  bytes of weak checksum is available. 
   INFO:   handleControlFile :  5  bytes of strong checksum is available. 
   INFO:   handleControlFile :  2  consecutive matches is needed. 
   INFO:   handleControlFile : zsync target file url is confirmed to be  QUrl("FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage") . 
   INFO:   handleControlFile : zsync target file sha1 hash is confirmed to be  "82FCF021EC6BCB163F721F68054503E423328A93" . 
   INFO:   handleControlFile : zsync target file has  181634  number of blocks. 
start with update process
   INFO:   setConfiguration : creating temporary file. 
   INFO:   setConfiguration : temporary file will temporarily reside at  "/home//Applications/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage.gZPhUnLKbH.part" . 
   INFO:   start : starting delta writer. 
   INFO:   verifyAndConstructTargetFile : calculating sha1 hash on temporary target file.  
   INFO:   verifyAndConstructTargetFile : comparing temporary target file sha1 hash( "82FCF021EC6BCB163F721F68054503E423328A93" ) and remote target file sha1 hash( "82FCF021EC6BCB163F721F68054503E423328A93" 
   INFO:   verifyAndConstructTargetFile : sha1 hash matches! 
   INFO:   verifyAndConstructTargetFile : file with target file name exists. 
   INFO:   verifyAndConstructTargetFile : renaming new version as  "FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2-revised-on-2021-01-30T11-36-46.12-x86_64.AppImage" . 

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0129.23662 +3166 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: bd2d90d312b3519573a323f97d7544d82f0e314d
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

snelweg, 2021-01-30:
Update:

I see that the new files are downloaded, it looks like AppImage just doesn't use them.

> ton@debian:~/Applications$ ls
balenaEtcher-1.5.113-ia32_5fb9a3ec0f06aece3ae4d56026f4dc91.AppImage
balenaEtcher-1.5.113-x64_f17aa034fe3f255bcf43f8665107a16c.AppImage
Cura-4.4.0_2925d1ab876b75cf407463405eb0c354.AppImage
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210129-glibc2.12-x86_64_39ebd499342219ec96fd253bc3c472af.AppImage
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2.12-x86_64.AppImage
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2-revised-on-2021-01-30T11-14-51.12-x86_64.AppImage
FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210130-glibc2-revised-on-2021-01-30T11-36-46.12-x86_64.AppImage
shotcut-linux-x86_64-201128_301bf65d33de36f86f3558b886ec2a96.AppImage
Ultimaker_Cura-4.8.0_6782abb3abec599bf7274297cdaa94bb.AppImage
webcamoid-8.7.1-x86_64_1d6d507a1ed8d65ca3a0710d3617c164.AppImage
ton@debian:~/Applications$

realthunder, 2021-01-30:
No idea what's going on there. Can you update the image using FC's own AppImage updater inside the preference dialog?

snelweg, 2021-01-30:
> No idea what's going on there. Can you update the image using FC's own AppImage updater inside the preference dialog?

That's what I did, sorry if I wasn't clear on that.
I used the internal AppImage updater from preferences->AppImage. In first post I copied all the messages the updater gives in the preferences->updater window.

The updater does download the latest release but it does not use it. I have to manually start the appimage to have it recognized by the appimage manager on Debian.

realthunder, 2021-01-30:
So is the downloaded one working now. Seems like a problem of the AppImage updater. And you still can't see the AppImage on github page?

snelweg, 2021-01-30:
> So is the downloaded one working now. 

Yes, after I start it up from the cli and AppImage inserts it into the menu I can use it. AppImage acts strangely anyway, not my favorite tool at all.

> Seems like a problem of the AppImage updater. 

Yes, I agree. 

> still can't see the AppImage on github page?

Yes, it does show now. Maybe I was too quick before?

closed 2021-01-30 by snelweg