realthunder/FreeCAD_assembly3 #442: dynamic solver not working

kth001, 2021-01-31:
in the assembly 3 wiki, it says there is a dynamic solver in asm3 but why it doesn't work? error message is 
parts still touched after recompute. in <app> document.cpp in line 3524 i don't know whether it's the problem of freecad or asm3 
i am posting this issue here because dynamic solver is function of asm3 i can do programming a little bit, so i may be solve this problem on my own but i absolutely don't know what and where is a problem p.s. my mother tongue is not an english so sorry for my poor english i hope you can understand

ceremcem, 2021-01-31:
It works only if you move the "master" part:

https://user-images.githubusercontent.com/6639874/106381133-dfd74180-63c7-11eb-99be-44b6ea799a55.mp4

As far as I could understand, first assembly is selected as master. 

### Attachments 

[axis-wheel.FCStd.remove-me.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5899530/axis-wheel.FCStd.remove-me.zip)

### Version

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26832 (Git)
Build type: Release
Branch: LinkStage3
Hash: d7b79274aab23bef2132b1094f222fd34104ffb8
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2021-01-31:
Here is how you can change the "master part": 

https://user-images.githubusercontent.com/6639874/106381327-f631cd00-63c8-11eb-9741-48a78782da82.mp4

kth001, 2021-02-02:
thanks for your answer but it doesn't helped my problem is not same as your video 
in my problem, it anyway moves, but it moves as if it doesn't have constraints

https://user-images.githubusercontent.com/36230303/106543252-aaa62d00-6548-11eb-912e-cad16fcabbfe.mp4

ceremcem, 2021-02-02:
Ah, you probably need to turn on the "Auto recompute" (and preferably "Smart recompute"):

[image]

kth001, 2021-02-02:
as in the video, it turned on

ceremcem, 2021-02-02:
Ah, yes, sorry. It is seen in full screen. Then I run out of ideas.

realthunder, 2021-02-02:
Can you please move those parts as shown in your own video and save the the log output in your report view and attach it here? I need to see the full error message. Also, please attach your FreeCAD About information here.

kth001, 2021-02-02:
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23463 (Git)
Build type: Release
Branch: master
Hash: adc6db8f32c17e9f67cb82aa7962d1e42ce061c6
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: Korean/South Korea (ko_KR)

20:05:51  1e-07 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.0390967 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.049971 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.0762248 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.0857601 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.111537 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.133992 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.160921 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.170614 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.196337 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.20586 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.231044 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.254639 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.278791 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.301852 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.327725 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.350542 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.382159 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.415168 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.438924 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.470699 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
20:05:51  0.494364 <App> Document.cpp(3524): axis_wheel#Parts still touched after recompute
this is full error message

grntls, 2021-05-19:
Hi,
I experienced the same problem as described here... and was waiting for it to be fixed.... but finally discovered that it is a side effect of the commit 9e34ecd1c05c12d2d5014d17ffccd7a6dfb14e31 
"mover: use SHIFT key to bypass recompute on moving"
... as my preferred control setting is "OpenInventor", for which SHIFT key is used to move objects, hence the problem...
For now just reverted this commit locally and it works.
Best regards

kth001, 2021-05-21:
oh thanks a lot. my control setting in openinventor too, and that was the source of the problems.
even in the released 0.19.1 version, this problem exists.
so, i am now changing control method while using dynamic solver. and it works! hooray!! but still i have a question cause i'm didn't use SHIFT key, i used CTRL key. but why this happens? and how did you find that a specific comment was the problem?

grntls, 2021-05-21:
As there was only very few people complaining about this problem I suspected something relative to my configuration (tried 3 versions of Freecad, on Win7, Win10, Linux... nothing helped). I had also tried Assembler 3 about 2 years ago and it was working (but not used it since because my simple projects didn't need it, which changes now). So after a few months I decided to review the commits on github and found the previous thanks to it's label.
Note that's it's only two lines of Python code that can easily be commented directly in the mod/freecad_assembly3/freecad/asm3/mover.py file of your installed version without recompiling anything, till it's fixed.

Concerning the CTRL key I didn't noticed it but you're true, and reverting the 9e34ecd1c05c12d2d5014d17ffccd7a6dfb14e31 commit does not help - probably comes from somewhere else. Moreover, I discover that I'm probably wrongly using the Shift Key as it is the Ctrl key that is documented in Freecad... I will not have time to investigate further now, but really appreciate that this vital function of Assembler 3 is working.

commit b998108c (2021-09-01): mover: ignore 'Shift/Ctrl' modifier if using inventor navigation

Fixes #442

@xcasxcursex @kth001
closed 2021-09-01 by realthunder