realthunder/FreeCAD_assembly3 #443: Manual sync or recompute of a Sub-ShapeBinder binded to a sketch is needed

Rabbit-sk, 2021-02-02:
Dear RealThunder,

This problem is quite annoying because of the need to manually sync many Sub-ShapeBiders.

https://user-images.githubusercontent.com/74265320/106579708-49c72680-6541-11eb-9fee-b06dc4c420c2.mp4

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0130.23662 +3170 (Git)
Build type: Release
Branch: LinkStage3
Hash: d7b79274aab23bef2132b1094f222fd34104ffb8
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2021-02-04:
I can't reproduce the problem here. Did you somehow set the binder's `BindMode` to `Frozen`? Can you please attach a file showing the problem?

Rabbit-sk, 2021-02-04:
That is strange, I also can't reproduce the problem 😃 This problem appears during my work, so I reproduce it in a new project and posted it here. Now it is fortunately not possible to reproduce it in a new project. But this problem still exists in the project files where the problem appeared. Here is the project file used for this report, where the problem is still present:
[Sub-ShapeBinder_To_Sketch_NoAutoSync.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5924390/Sub-ShapeBinder_To_Sketch_NoAutoSync.zip)
Password: RealThunder_came__RealThunder_saw__RealThunder_fixed

ceremcem, 2021-02-04:
(Why password? Why :D)

Rabbit-sk, 2021-02-04:
Just for fun! 😃 And also to praise RealThunder's work.

ceremcem, 2021-02-04:
That's a recomputation problem. Right click `Body`, `-> Recompute object` and problem goes away.

ceremcem, 2021-02-04:
That's a recomputation problem. Right click `Body`, `-> Recompute object` (and manually click the "Recompute" button on the top, it works like "mark to recompute" at the moment). After recomputation, problem goes away.

Rabbit-sk, 2021-02-04:
Not as easy. I must do right click and Recompute on Sub-ShapeBinder. Then it is correct, but after another change of the Sketch in the first Body, the mannual recompute is needed again. In project where are a lot of such Sub-ShapeBinders it is an issue, as one sometimes forget which all need to be recomputed. As I wrote, fortunately it is not occuring again, but what with the already nonsyncing objects in existing projects? Maybe rework, I do not know.

Rabbit-sk, 2021-02-04:
Interesting finding: If Body which contains sketches is moved in some direction, everything is recomputed, and another later changes to the sketches are also synced. Also if mentioned Body is moved to original position.

ceremcem, 2021-02-04:
> Not as easy.

I didn't mean to provide a workaround, I mean I did confirm. Sorry for misunderstanding.

>  Maybe rework, 

Does not sound okay. We heavily rely on subshape binders and it literally (obviously) affects the actual production. Such a rework approach would end up with human errors.

Rabbit-sk, 2021-02-04:
> I didn't mean to provide a workaround, I mean I did confirm. Sorry for misunderstanding.

That is okay.

> We heavily rely on subshape binders

Yes, very heavily. Sometimes I have in a project such Sub-ShapeBinders spaghetti, that reworking some early created object would necesairly lead to freeze or detach later Sub-ShapeBinders. Not mentioning that searching of all Sub-ShapeBinders relations to faces and sketches of particular body. Sometimes corrections of earlier sketches are small and it is not visible at the first sight, that model was not recomputed.

Rabbit-sk, 2021-02-04:
> **Interesting finding:** If Body which contains sketches is moved in some direction, everything is recomputed, and another later changes to the sketches are also synced. Also if mentioned Body is moved to original position. 

But I also observed that if body is moved for example 100 mm in some edirection and confirming it, Sub-ShapeBinders and dependent shapes were recomputed, but moved object was still on its original position in 3D view. After changing position to previous one in Data tab was fixing it.

It seems that this issue has relatively easy fix, but it is needed to check whole project if something that should sync was not synced after some change in a sketch or shape used for some Sub-ShapeBinder.

closed 2021-02-07 by Rabbit-sk