realthunder/FreeCAD_assembly3 #445: TreeView don't scroll to higlighted item if Down or Up key is used, or to new item like non-solid feature

Rabbit-sk, 2021-02-02:
Dear RealThunder,

It would be more conveniet if position of the vertical scroll bar would be adjusted to show item higlighted by `Up` or `Down` keys, and also to show all new features in the body.

https://user-images.githubusercontent.com/74265320/106590665-a4ff1600-654d-11eb-839b-61b4258a3796.mp4

https://user-images.githubusercontent.com/74265320/106590175-1e4a3900-654d-11eb-9f0c-39115e8eae0b.mp4

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0130.23662 +3170 (Git)
Build type: Release
Branch: LinkStage3
Hash: d7b79274aab23bef2132b1094f222fd34104ffb8
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2021-02-03:
You disabled tree view `Sync selection` option, right? You can use `T, G` shortcut to scroll to the current selection manually.

Rabbit-sk, 2021-02-04:
> You disabled tree view `Sync selection` option, right?

Yes, it depends on what I'm doing. I toggle this setting sometimes. BTW I found out that SyncSelection does not have impact on this behavior everytime. I'll describe it below.

> You can use `T, G` shortcut to scroll to the current selection manually.

Thank you for reminding me. But with `SyncSelection` turned off some new items are focused and some not. Maybe they all should be treated the same. I'll also describe it below.

I found out that things which I reported in this issue should probably be divided into two issues. You have already commented on it, so I will write about both of them in this thread, because a significant change to my first comment and title would not be appropriate.

- TreeView don't scroll to higlighted item if `Down` or `Up` key is used after an item was moved to a part container by drag and drop with mouse or `T, D` (regardless of `SyncSelection` setting):

https://user-images.githubusercontent.com/74265320/106888642-26d67700-66e7-11eb-98be-abc489c1aafb.mp4

- With `SyncSelection` turned off TreeView don't scroll to some new items in a body (Datum Plane and Sub-ShapeBinder):

https://user-images.githubusercontent.com/74265320/106590665-a4ff1600-654d-11eb-839b-61b4258a3796.mp4

In principle, it is similar to this from a UX point of view: [If ... moved outside of the current view before ...](https://github.com/realthunder/FreeCAD_assembly3/issues/415#issuecomment-771728446)