realthunder/FreeCAD_assembly3 #452: Not a bug >> just an idea for you realthunder

FreeCADAntiBugs, 2021-02-04:
Good morning :)
this is an idea it could get you more fund and respected FreeCAD without scum int it.
make a new branch and call it "FreeCAD Pro"
in this branch get rid of all unwanted modules except
- Sketch
- Part (& part design merged together)
- TechDraw
- Path
- Assembly3 (or Assembly)
- FEM
- Rendering
- you can add important functions to the suggested modules above,
then stop any development except maintaining the current one's
i don't know what WB modules written in C++ but if most of those written in it
that will be enough for us.
this will make FreeCAD Pro more reliable, dependent and easier to maintain.
specially if you added ribbon toolbar just like autodesk inventor.
think about that .. i can help you, i don't need money or donnation  :)
if you accept it i'll make a new wordpress site with a nice theme for the new branch as a gift + all graphics work (icons, logos etc) for the new born freecad pro :)
thank you.

realthunder, 2021-02-05:
Thanks for the suggestion. I really wish FreeCAD can one day be considered as 'Pro'. But today is not that day, not when you compare it with most (if not all) mid-range commercial software out there. I can, and I am giving it a big push so that maybe we can see that day come a bit earlier, but let's face it, one man's effort is not nearly enough.

I don't see anything wrong with today's FreeCAD community. On the contrary, I think it's openness has produced many wonderful and innovative external modules. It really should be considered as its strength. On the other hand, I do see that the lack of an organization is what's holding it back for all those years. And the community knows that, and they are taking actions for this. I am no manager material, but I have expressed my willingness to be a full time developer for FreeCAD if there is funding. Let's just hope that we can see some light on this soon. Meanwhile, it's not like that I will be just sitting there and waiting for it to happen. And you can propose you ideas just the same, but please, let's take one step at a time.

ceremcem, 2021-02-05:
@FreeCADAntiBugs Related: [Making projects resistant to breaking changes](https://forum.freecadweb.org/viewtopic.php?f=8&t=54298)

FreeCADAntiBugs, 2021-02-05:
Okay dear @realthunder as you wish :)
i shall continue exposing bugs in FreeCAD with you.
thank you.

FreeCADAntiBugs, 2021-02-05:
> @FreeCADAntiBugs Related: [Making projects resistant to breaking changes](https://forum.freecadweb.org/viewtopic.php?f=8&t=54298)
Hello @ceremcem,
what made me write this is
- no commitment or rules for developers just anyone at any place
- no hierarchical
- no structure
- no imposed design
- no naming
- no, no etc ...
what made Blender successful is that the main program are integrated with a solid design structure, no one can break the rules.
even if they decided to include addon inside main body, they certify it first.
thank you.

closed 2021-02-05 by FreeCADAntiBugs