realthunder/FreeCAD_assembly3 #453: Tiny 3D cube on 5K display

irmantus, 2021-02-05:
That 3D cube thingamajig is very tiny on iMac 5K retina display, imposible to use.

[image]

ceremcem, 2021-02-21:
@irmantus Can you check if that issue is resolved in https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11?

irmantus, 2021-02-22:
By default it was still small, but now I see size settings so I set to 256 and it looks nice.
[image]

closed 2021-02-22 by ceremcem