realthunder/FreeCAD_assembly3 #455: [Feature Request]

bf5man, 2021-02-06:
**Context**: In Part Design, to create a sketch on a face, the user first needs to select the face he wants to have the sketch on and then clicks the "Create a new sketch" button.

**The issue**: I often find myself forgetting to select a face prior to clicking the "Create a new sketch" button. When it happens, I need to close the feature since that feature only allows you to select one of the three datums AFAIK.

**The proposed improvement**: If the user doesn't select a face prior to clicking the "Create a new sketch" button, then he would be allowed to select any arbitrary flat face in the 3D view. Then, clicking the "OK" button would initiate a sketch on that selected face.

Thanks,

ceremcem, 2021-02-06:
Would a "No face is selected. Use datum planes? (Yes/No)" popup help developing the correct habit?

Rabbit-sk, 2021-02-06:
> a "No face is selected. Use datum planes? (Yes/No)" popup

It would be better without such pop-up. Imagine a experienced user, already used to the logic of using FreeCAD, who want to create a sketch on one of the datum planes of active body. This pop-up would introduce extra movement of the mouse and click - an annoying delay.

ceremcem, 2021-02-06:
>  Imagine an experienced user, already used to the logic of using FreeCAD, who want to create a sketch on one of the datum planes of active body, without expanding its Origin branch and selecting a particular datum plane. This pop-up would introduce extra movement of the mouse and click - an annoying delay.

Exactly. I thought the same thing, hence the proposal. realthunder would surely bind it to a Preferences option, so an experienced user wouldn't need to see it more than once. The point of the second proposal is that it is easily overridable (in terms of FreeCAD development time). 

However, the original proposal introduces whole new UX which can not be overridden easily. Existing users would need to deal with a whole new UX. It doesn't explain how it will handle the current functionality (ie. creating sketches on datum planes). 

(PS: I never create sketches on datum planes, I just try to empathize.)

bf5man, 2021-02-06:
I think there might be a slight misunderstanding with my proposal. 

What I propose is not to add a popup or a new dialog window.

I propose that if "create a new sketch" is clicked without a face already selected, it would still launch the selection feature as usual. But **then**, it would let the user select a face in the 3D space. That wouldn't prevent anyone selecting a datum on the left in the "select feature" window (current behavior). What I envision would be that if I were to select a face in the 3D window, that selected face would be automatically added to the "select feature" box to the left and be selected. Should it be complemented with a check box similar to the "Allow used feature" below the "select feature" field which might be named "Allow direct face selection"? Maybe, but I'm not sure it would bring clarity or any benefit.

So in theory, it doesn't change the tool workflow, that's only remediating the fact that only datum faces are currently selectable at that "create new sketch" step.

In my opinion, the current behavior is inconsistant on a user standpoint. If I select a face first and then select the "create a new sketch" feature it automatically works and brings me to the sketcher. Forgetting to select a face brings you in an interface where you select only select datums, even if 3D faces are still selectable on the 3D window **and** its perfectly correct and possible to place a sketch on a face anyway. The button is not named "create new sketch on a datum", so why limit it to only datums?

You might reply to that "_the current UX in Freecad is to select a feature/geometry first and then select the functionality you what to apply to it then_". But at the same time, compare that with the UX for placing dimensions in the sketcher, you either select the edge/points first and then select the dimension functionality and a dimension pops up **or** you can select the functionality first and then select the edge/points and the dimension is placed. Either UX are "correct" per se and the user doesn't have to face unecessary frustrations, it just "work". That's what I intend to suggest with my feature request here.

ceremcem, 2021-02-06:
> In my opinion, the current behavior is inconsistant on a user standpoint. If I select a face first and then select the "create a new sketch" feature it automatically works and brings me to the [...]

You have a point here. I agree.

realthunder, 2021-02-13:
I have made some major changes to sketch creation in `PartDesign`. It will now bring up the sketch attachment editor if no support is selected prior to creating the sketch. The attachment editor supports selecting all kinds of the supporting object, not limited to face.

In addition, `PartDesign` and Sketcher will now auto handle external object reference using `SubShapeBinder`. This means that you can select any arbitrary object as support or sketch external geometry, regardless of where the object is located.

ceremcem, 2021-02-14:
Nicely works as of https://github.com/realthunder/FreeCAD/commit/8483f93. Thanks.

bf5man, 2021-03-01:
Nicely implemented RealThunder, thanks!

closed 2021-03-01 by bf5man