realthunder/FreeCAD_assembly3 #457: Sketch: XML Attribute: 'internalGeometryType' not found

Manu-Fr, 2021-02-06:
I encountered a problem with the current version (2021.02.05) when I loaded a freecad file created with an **earlier version** (I don't remember exactly which one). The error message was as follows : 

**XML Attribute: 'internalGeometryType' not found In context: |xxx|Sketch001|Geometry.**

I try to repair the sketch but it leads to an empty sketch ....

I created a new simple sketch with the current version of Freecad_assembly3. I unzipped the FCStd file, edited the Document.XML and found this for all geometries : 

`<GeoExtensions count="1">
                <GeoExtension type="Sketcher::SketchGeometryExtension" id="4" internalGeometryType="0" geometryModeFlags="00000000000000000000000000000000"/>
              </GeoExtensions>`

So, I edited the FCStd file which was not working and found : 

`<GeoExtensions count="1">
               <GeoExtension type="Sketcher::SketchGeometryExtension" id="171"/>
</GeoExtensions>`

I guess the current version expects to have more information in this "GeoExtensions" section, and fails if missing.
So I cleared the "GeoExtensions" section for all geometries of my file, reloaded and now it's working with the current version. I saved and re-edited the file, I could notice the "GeoExtensions" sections were all created with the missing information.

I think I solved my problem but I don't know if this is the best way...

Thanks

Best regards

realthunder, 2021-02-06:
Can you please attach the problematic file here? Or, you can send it to my email realthunder.dev at gmail.

luzpaz, 2021-07-13:
@Manu-Fr ping

closed 2021-08-04 by luzpaz