realthunder/FreeCAD_assembly3 #459: 1 mm aded to position of link instances

Rabbit-sk, 2021-02-08:
Dear RealThunder, 

please look at x coordinate of these link objects, which have cumulating addition to their position:
(The y coordinate was already changed by me.)
[image]
[image]
[image]
[image]
[image]
And so on...

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2021-02-10:
The increment is on purpose, for distinguish of different array element. It is probably better to use the shape boundary for better choice of increatment.

Rabbit-sk, 2021-02-10:
> The increment is on purpose, for distinguish of different array element.

Makes sense. If an object is not much bigger than 1 mm in any direction, then it is distinguishable:
[image]

But if it has at least one dimension much bigger than 1 mm, these offsets are not noticable, only after longer zooming and in the Data tab. Ofcourse one can easily change this increment in position to whathever desired value. With bigger objects, fitted to the view, it still seems that they are all in one place and undistinguishable. So for me it seemed as a bug. 

Increment of position to distinguish array elements is good idea, but 1 mm is not enough for all purposes that FreeCAD can be used. Would it be possible to obtain the size of the bounding box of linked object (body or part container) and use that value to increment array elements?
[image]

> It is probably better to use the shape boundary for better choice of increatment.

Is this what I decribed above?

realthunder, 2021-02-14:
> Is this what I decribed above?

Yes, that's right. And the above commit implemented just that.

Rabbit-sk, 2021-02-23:
It works better now. There is only a weird placement pattern if bigger number of instances is entered at once:

https://user-images.githubusercontent.com/74265320/108812081-bff20280-75ae-11eb-9f60-3c9448bb8493.mp4

Verified in the build compiled from the source code after commit [9deefac](https://github.com/realthunder/FreeCAD/commit/9deefac4f2748d4af44d5d856f605c92d736c8c3).

Rabbit-sk, 2021-02-25:
> There is only a weird placement pattern if bigger number of instances is typed (in video from 0:34 to 0:45)

Since no object overlaps with another, and instances are later placed in the desired position, this somewhat strange initial location is irrelevant. I consider this issue solved.

closed 2021-02-25 by Rabbit-sk