realthunder/FreeCAD_assembly3 #460: Python functions in expressions sometimes cause "Access denied of callable" errors

grggnzlz, 2021-02-12:
Hi RealThunder.  I started using python functions in expressions because they seemed to work ok when I added them with setExpression().  Now I'm getting some issues like the one below.  The behavior isn't consistent.  Sometime I get the errors, and sometimes I don't.  Sometimes they go away with a recompute.  I can reproduce the problem in the attached by opening the script named "Profile" and then closing it.  When I do that a bunch of bodies show the error in the tree view.  I can make them go away by recomputing the LeverPlate object.  Thanks for all the great work on this.

```
15:39:46  <Exception> Access denied of callable in unknown module
in expression: <<Reeds>>.Geometry[135].EndPoint.sub(<<Reeds>>.Geometry[135].StartPoint).y
in binding '.Placement.Rotation.Angle'
```

```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: English/United States (en_US)
```

[debug2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5974440/debug2.zip)

realthunder, 2021-02-15:
This is indeed a bug. The above referenced commit should fix it.

On the other hand, there are a few problems with your expressions. `Base.Vector` supports Python numeric protocol, meaning that they can directly subtract each other without using a callable, and FC expression can do the same. It is advised to not use callable whenever you can because of security concern. The bug you discovered is in the added logic for addressing that concern, and is currently not matured yet.

The other problem is related to the way you reference sketch geometry. Once you delete any geometry beyond the referenced one (i.e. geometry with a smaller index), you'll get an error. The correct way to reference any geometry inside the expression is through the subname reference syntax described [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#user-content-subobject). With this syntax, the geometry element reference is internally tracked, and will auto correct itself on changes in sketch.

To find out the geometry element reference, simply hover your mouse above the geometry while editing the sketch, and find it in the status bar message, as shown below,

[image]

Notice the small case `vertex` instead of the usual Capital `Vertex`. This is for sketch internal references, which can be used to refer to normal, construction, or external geometry.

So in your case, the preferable expression should be something like this,

```python
(<<Reeds>>.<<.vertex271>>._shape.Point - <<Reeds>>.<<.vertex270>>._shape.Point).y
```

ceremcem, 2021-02-19:
Very same error is displayed while opening my old design with a recent FreeCAD version: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27228 (Git)
Build type: Release
Branch: LinkStage3
Hash: 99ccabfcf730689dfe1d027d90addd11332f193b
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

This error doesn't exist in `f7007966e1f73a423bef124c114587768f655609` and *does exist* in `4567b5ad29a39121088202a5575cebc6640c42c9` (my previous snapshot).

grggnzlz, 2021-02-19:
Thanks.  Looking forward to the build with this fix.  Good to know about Base.Vector and the benefits subnames.

ceremcem, 2021-02-19:
Current version (LinkStage3 @ 11256419507383c7006b2fbc1cf558158acc7230 and counterpart FreeCAD_Assembly3 version) does not solve my case. @realthunder I'm sending my model privately to your Gmail address.

ceremcem, 2021-02-19:
FYI: Above commit fixed the errors.

luzpaz, 2022-01-12:
@grggnzlz please retest on latest Daily and report back. Thanks!

closed 2022-07-11 by luzpaz