realthunder/FreeCAD_assembly3 #462: Segmentation fault when attempting to initiate a

bgamari, 2021-02-13:
# Reproducing

1. Open the attached test document ([test.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/5976172/test.tar.gz))
2. Enter the Assembly 3 workbench
3. Select `Body`
4. Click on the "Axial move part" button
5. Observe that FreeCAD crashes

# Versions

 * https://github.com/realthunder/FreeCAD/commit/03ccccbf0f3c4efa19d5dd110136054930fd6c39
 * https://github.com/realthunder/FreeCAD_assembly3/commit/da4b497b3f48f9ae34de6a3fca0249c96832f1c2
 * https://github.com/realthunder/coin/commit/5dd4b9854bc4d5c8bbd89093204c7702d8dcc99f

# Backtrace

```
$ FreeCAD
FreeCAD 0.19, Libs: 0.19RUnknown
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
si_init_perfcounters: max_sh_per_se = 2 not supported (inaccurate performance counters)
A3 available
DeprecationWarning: Second argument is deprecated. It is ignored and will be removed in future versions. The default Python feature proxy is used for extension method overrides.
Traceback (most recent call last):
  File "/nix/store/sa9dpws3mifbkh166m5fr1whvwvq2b7p-freecad-asm3-plugin/Ext/freecad/asm3/assembly.py", line 4024, in getConstraintGroup
    ret = obj.Group[0]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

DeprecationWarning: Second argument is deprecated. It is ignored and will be removed in future versions. The default Python feature proxy is used for extension method overrides.
Traceback (most recent call last):
  File "/nix/store/sa9dpws3mifbkh166m5fr1whvwvq2b7p-freecad-asm3-plugin/Ext/freecad/asm3/assembly.py", line 4074, in getElementGroup
    ret = obj.Group[1]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

DeprecationWarning: Second argument is deprecated. It is ignored and will be removed in future versions. The default Python feature proxy is used for extension method overrides.
Traceback (most recent call last):
  File "/nix/store/sa9dpws3mifbkh166m5fr1whvwvq2b7p-freecad-asm3-plugin/Ext/freecad/asm3/assembly.py", line 4096, in getPartGroup
    ret = obj.Group[2]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

DeprecationWarning: Second argument is deprecated. It is ignored and will be removed in future versions. The default Python feature proxy is used for extension method overrides.
75.871426 <asm3.cstr> constraint.py(559): no fixed part
Program received signal SIGSEGV, Segmentation fault.
#0  /nix/store/a3syww9igm49zdzq3ibzw9m8ccvsgxla-glibc-2.32/lib/libc.so.6(+0x38480) [0x7fcbd684d480]
#1  0x7fcbd8d4ed70 in SoSFInt32::setValue(int) from /nix/store/5wwd0lbs5gan5pimdavzg3y371rln469-coin-4.0.0/lib/libCoin.so.80+0
#2  0x7fcbdad6425d in Gui::SoFCCSysDragger::SoFCCSysDragger() from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0x312d
#3  0x7fcbdaea8b6b in Gui::ViewProviderLink::startEditing(int) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0xbb
#4  0x7fcbdaa44834 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0xaa4
#5  0x7fcbdaa6b2e1 in Gui::DocumentPy::setEdit(_object*) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0xd1
#6  0x7fcbdaa6b495 in Gui::DocumentPy::staticCallback_setEdit(_object*, _object*) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0x55
#7  /nix/store/fzgk230riagc58n8w1fa41ngs723pzpb-python3-3.8.6/lib/libpython3.8.so.1.0(PyCFunction_Call+0xf3) [0x7fcbd941dbf3]
#8  /nix/store/fzgk230riagc58n8w1fa41ngs723pzpb-python3-3.8.6/lib/libpython3.8.so.1.0(_PyObject_MakeTpCall+0x87) [0x7fcbd941cd47]
#9  /nix/store/fzgk230riagc58n8w1fa41ngs723pzpb-python3-3.8.6/lib/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x4cf0) [0x7fcbd948aaf0]
[... 19 lines cut ...]
#29  0x7fcbd8055da3 in QApplication::notify(QObject*, QEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5+0x583
#30  0x7fcbdaa9b988 in Gui::GUIApplication::notify(QObject*, QEvent*) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0x88
#31  0x7fcbd758f7fa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Core.so.5+0x12a
#32  0x7fcbd8054eb7 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5+0x1b7
#33  /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5(+0x1cfe19) [0x7fcbd80a9e19]
#34  /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5(+0x1d3114) [0x7fcbd80ad114]
#35  0x7fcbd804ccef in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5+0x7f
#36  0x7fcbd8055b50 in QApplication::notify(QObject*, QEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Widgets.so.5+0x330
#37  0x7fcbdaa9b988 in Gui::GUIApplication::notify(QObject*, QEvent*) from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0x88
#38  0x7fcbd758f7fa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Core.so.5+0x12a
#39  0x7fcbd79ad013 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Gui.so.5+0x6c3
#40  0x7fcbd79ae6a5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Gui.so.5+0x255
#41  0x7fcbd7986dab in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Gui.so.5+0xbb
#42  /nix/store/rqj9v6ap9l6sr7cxparf9vdx4a62fnpi-qtbase-5.15.0-bin/lib/qt-5.15.0/plugins/platforms/../../../../../xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5XcbQpa.so.5(+0x6973a) [0x7fcbc108773a]
#43  /nix/store/rrk4d3mcp003z0kr8wg922ww3h1gfwq5-glib-2.66.3/lib/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fcbd3a57c4d]
#44  /nix/store/rrk4d3mcp003z0kr8wg922ww3h1gfwq5-glib-2.66.3/lib/libglib-2.0.so.0(+0x53ed0) [0x7fcbd3a57ed0]
#45  /nix/store/rrk4d3mcp003z0kr8wg922ww3h1gfwq5-glib-2.66.3/lib/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7fcbd3a57f5f]
#46  0x7fcbd75e86be in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Core.so.5+0x5e
#47  0x7fcbd758e38b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Core.so.5+0x12b
#48  0x7fcbd75965f0 in QCoreApplication::exec() from /nix/store/xxfcll4nvha28alliqnxf852y5f0pmx9-qtbase-5.15.0/lib/libQt5Core.so.5+0x90
#49  0x7fcbda9f9bb5 in Gui::Application::runApplication() from /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/lib/libFreeCADGui.so+0x3e5
#50  /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/bin/FreeCAD() [0x405930]
#51  /nix/store/a3syww9igm49zdzq3ibzw9m8ccvsgxla-glibc-2.32/lib/libc.so.6(__libc_start_main+0xed) [0x7fcbd6838dbd]
#52  /nix/store/kyal9cfaykvsmxa72ciig9iqzgdf6xn6-freecad-unstable-2020-10-17/bin/FreeCAD() [0x405c3a]
```

christian-nils, 2021-02-14:
I experience the same issue.

ceremcem, 2021-02-14:
I tracked it back a little bit, it still crashes in `f7007966e1f73a423bef124c114587768f655609`. However following version works fine: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26863 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

```

christian-nils, 2021-02-14:
Yeah, I can confirm on my machine too. It looks like the commit following https://github.com/realthunder/FreeCAD/commit/9391fcd19568d635bb290db09f9d616e6acdc4f4 breaks something.

christian-nils, 2021-02-14:
It is fixed now for me thanks to https://github.com/realthunder/FreeCAD/commit/08fdaf6a5f350130c21fea5732daaf02ec592efe.

Thank you @realthunder !

ceremcem, 2021-02-14:
Confirmed fix as of https://github.com/realthunder/FreeCAD/commit/8483f93. Thanks.

bgamari, 2021-02-14:
Indeed, I've confirmed that this has been fixed. I am now also seeing allocator `abort`s when saving, but this seems like an orthogonal issue. I'll open another ticket once I have characterised further.

closed 2021-02-14 by bgamari