realthunder/FreeCAD_assembly3 #463: extrude commant not working is  normal profile to curved surface

sedatkrn, 2021-02-14:
Freecad Linkstage did not make extrude normal sketch to cruved surface.
i made a cylinder on the frontplane and a creating a plane to top plane and offset from cylinder over de radius. for example ccylinder r=60mm h=100mm.i put the plane  top plane and giving to z offset about 80mm. create and sketc to offseting plane and create a circle r=20mm. 

i click to pad command for circle sketch. it is runing open the left pad menu. i choose to type box andi choosen up to the surface. i sellect to cylinder cruved surface but didnt make extrude. normaly freecad makes it. thanks...

ceremcem, 2021-02-14:
Can you please clarify your intention, I found it too dispersed to understand. Please provide a set of instructions to reproduce your intention.

sedatkrn, 2021-02-14:
Cerem bey iyi çalışmalar.  bir daire profili bir yüzeye kadar extrude yapmak istiyorum ancak yapacağım yüzey bir silindirin yan yüzeyi gibi eğri bir yüzey. Linkstage3  yalnızca düzlemsel yüzeylere kadar extrude yapıyor.  

> 
> 
> Can you please clarify your intention, I found it too dispersed to understand. Please provide a set of instructions to reproduce your intention.

ceremcem, 2021-02-14:
So you want to extrude a non-flat surface, like this: 

[image]

Is that correct?

sedatkrn, 2021-02-14:
bahsetmek istediğim bu. şu daire silindir yüzeyine extrude yapamıyorum.şu anda sanırım desteklemiyor. dosyayı ekledim. çok teşekürler ilginiz için.

This is what I want to talk about. I can't extrude that circle cylinder surface. I guess it doesn't support it right now. i uloded the file. tahanks for intrest.

[image]

https://github.com/sedatkrn/resimler/blob/main/extrude%20problem.FCStd

ceremcem, 2021-02-15:
This can be achieved by using `Type: Two dimensions` within the extrusion menu: 

[image]

Use `0 mm` as `Length` and an appropriate length (eg. 100mm) for `2nd length`: 

[image]

(You're welcome)

sedatkrn, 2021-02-15:
Ben bunu böyle yapılabileceğini biliyorum fakat demek istediğim eğri yüzeylere extrude  yapamamsı, ancak normal freecad süürümü bunu yapabiliyor. acaba bu eksik ne zaman giderilir.
I know it can be done this way, but I mean it can't extrude into curved surfaces, but the normal freecad version can do it. I wonder when this missing is fixed.

ceremcem, 2021-02-15:
> ancak normal freecad süürümü bunu yapabiliyor.

Eğri yüzeylerde böyle bir işlem yapılabildiğini bilmiyordum. (I didn't know that such a thing is even possible). 

Works as of above commit: 

https://user-images.githubusercontent.com/6639874/107939587-01275880-6f98-11eb-959a-8f85dc3f20c5.mp4

sedatkrn, 2021-02-15:
Çok faydalı bir özellik. It is so usefully tools.

closed 2021-03-07 by sedatkrn