realthunder/FreeCAD_assembly3 #464: FreeCAD crashes after deletion of multiple items in a sketch

Rabbit-sk, 2021-02-15:
Dear RealThunder, 

after selecting and deleting multiple items in a sketch, nothing dissapear, and after click, FreeCAD crasches:

https://user-images.githubusercontent.com/74265320/107926927-9c0f3b00-6f76-11eb-9acc-51a60e3773d6.mp4

Another time, when I moved mouse over geometry that should be deleted, in Report View in addition to those warnings that are in the video above, another warnings appeared:
[image]

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

ceremcem, 2021-02-19:
I could reproduce the crash. Note that it only crashes if you select from right-down to left-up and then press `Delete` key on the keyboard. 

I can also confirm that this crash does not happen in the recent version: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27235 (Git)
Build type: Release
Branch: LinkStage3
Hash: 23bafae49eb9858e1dde35e58f2f0ebca76a8f0a
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

oficinerobotica, 2021-06-09:
Tested with:
```
OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
'''
and the issue isn't present anymore.

closed 2021-06-09 by oficinerobotica