realthunder/FreeCAD_assembly3 #465: FC crashes when I try to copy a formula

snelweg, 2021-02-15:
When I try to copy a 'formula' from the value box FC crashes. It happens all the time and is getting really annoying. I do not want ti use the spreadsheet and use the values from the Sketches themselves. This always works great, but now I have big problems with it.

[image]

https://user-images.githubusercontent.com/1257901/107964688-d6de9680-6fa9-11eb-986f-34651b7afe9b.mp4

[z-as.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5982463/z-as.zip)

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

ceremcem, 2021-02-15:
I can confirm, crash can be reproduced in: 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27217 (Git)
Build type: Release
Branch: LinkStage3
Hash: 4567b5ad29a39121088202a5575cebc6640c42c9
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

### Backtrace 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7fce1c353840]
#1  [0x5578ac3395a0]
```

snelweg, 2021-02-15:
Same here indeed:

> debian:~/Applications$ ./FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210205-glibc2.12-x86_64_e991b3b66860f1b1456242071ae5d7a3.AppImage 
/tmp/.mount_FreeCANRBCux
FreeCAD 0.19, Libs: 2021.0205R23662 +3201 (Git)
Sheet Metal workbench loaded
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x37840) [0x7f7a8df1e840]
#1  [0x555c617b2720]

sigxcpu76, 2021-02-15:
```Thread 1 "FreeCADLink" received signal SIGSEGV, Segmentation fault.
0x00007fab043166f8 in vtable for QPaintDevice () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
(gdb) bt
#0  0x00007fab043166f8 in vtable for QPaintDevice () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
#1  0x00007fab059336a5 in Gui::ExpressionTextEdit::contextMenuEvent(QContextMenuEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#2  0x00007fab044ca9e2 in QWidget::event(QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#3  0x00007fab04533db7 in QFrame::event(QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#4  0x00007fab03c77019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#5  0x00007fab0449f7e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#6  0x00007fab044a469d in QApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#7  0x00007fab059489f9 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#8  0x00007fab03c79606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#9  0x00007fab044da094 in ?? () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#10 0x00007fab044daf7c in ?? () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#11 0x00007fab0449f7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#12 0x00007fab044a531d in QApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#13 0x00007fab059489f9 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#14 0x00007fab03c79606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#15 0x00007fab0401dec5 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
#16 0x00007fab0401e91f in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
   from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
#17 0x00007fab0400bc65 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
#18 0x00007faafab75010 in ?? () from /tmp/.mount_FreeCAFiYFnI/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5
#19 0x00007fab003107dd in g_main_context_dispatch () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/./libglib-2.0.so.0
#20 0x00007fab00310a71 in g_main_context_iterate.isra () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/./libglib-2.0.so.0
#21 0x00007fab00310b11 in g_main_context_iteration () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/./libglib-2.0.so.0
#22 0x00007fab03ca80e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#23 0x00007fab03c76019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#24 0x00007fab045e08d9 in QDialog::exec() () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#25 0x00007faa68a736a8 in CmdSketcherConstrainRadius::activated(int) () from /tmp/.mount_FreeCAFiYFnI/usr/lib64/SketcherGui.so
#26 0x00007fab05998724 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#27 0x00007faa68a43537 in CmdSketcherCompConstrainRadDia::activated(int) () from /tmp/.mount_FreeCAFiYFnI/usr/lib64/SketcherGui.so
#28 0x00007fab059984c0 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#29 0x00007fab0596562c in Gui::ActionGroup::onActivated() () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#30 0x00007fab03c8fa18 in QMetaObject::activate(QObject*, int, int, void**) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#31 0x00007fab0449a928 in QAction::triggered(bool) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#32 0x00007fab0449c2b2 in QAction::activate(QAction::ActionEvent) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#33 0x00007fab0449ca3b in QAction::event(QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#34 0x00007fab0449f7f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#35 0x00007fab044a531d in QApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Widgets.so.5
#36 0x00007fab059489f9 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#37 0x00007fab03c79606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#38 0x00007fab04035b51 in QShortcutMap::dispatchEvent(QKeyEvent*) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Gui.so.5
[... cut ...]
#48 0x00007fab03ca80e8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#49 0x00007fab03c76019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#50 0x00007fab03c79e8f in QCoreApplication::exec() () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib/libQt5Core.so.5
#51 0x00007fab058c3d6e in Gui::Application::runApplication() () from /tmp/.mount_FreeCAFiYFnI/usr/bin/../lib64/libFreeCADGui.so
#52 0x0000560f6ef6d4bf in main ()
```

ceremcem, 2021-02-20:
I can confirm that this is fixed in recent version. Should we close it?

snelweg, 2021-02-21:
@ceremcem I use the AppImage, there is no new version so I can not confirm it is fixed.

luzpaz, 2021-02-21:
@ceremcem maybe we should mention in the Assembl3/LinkStage3 how to build from source for those who are adventerous? You have a repo that explains this, right?

ceremcem, 2021-02-21:
[My repo](https://github.com/ceremcem/build-freecad-asm3) makes it trivial to build Assembly3 on Linux, yes. If anyone wants to try building Asm3 from source like AppImage (which you don't actually touch your host system at all), it's a good way to go.

snelweg, 2021-02-21:
@ceremcem @luzpaz Please do not make the same mistake the nerds at the official branch of FC made. There are loads of people who like me want to use beta software and are willing to report problems but have zero interest whatsoever to compile stuff themselves. I am a user and a maker, not a softwarebuilder. Testing is fun, using the latest and greatest too. But my fun stops at compiling. I'd rather spend my time using the software then compiling it.
I always thought that what the daily builds were for, a snapshot of the future. Not an invitation to build yourself, and why should one want to build themselves? I do not see the need for that at all, I will stick with the releases RT gives us.
Maybe the reviewing system can be streamlined and cases are only closed after a new release that deals with the problem has been released.

luzpaz, 2021-02-21:
Lol, now that isn't a nice thing to say about others is it? Also I don't get that impression that the folks in the master branch are like that.  

Anyway, I wasn't asking if you would compile. it was an off-topic question directed to ceremcem.  We are in agreement that compiling does suck. Apparently some dude on twitter says he has a https://github.com/realthunder/FreeCAD_assembly3/issues threadripper multicore machine that compiles FC in 10mins. That would be amazing.  But I digress...

snelweg, 2021-02-21:
> Lol, now that isn't a nice thing to say about others is it? Also I don't get that impression that the folks in the master branch are like that.

Well, it's my experience that they are. I do not use the forums for that reason anymore.

> Anyway, I wasn't asking if you would compile. 

Could have fooled me.

>it was an off-topic question directed to ceremcem. We are in agreement that compiling does suck. Apparently some dude on twitter says he has a threadripper multicore machine that compiles FC in 10mins. That would be amazing. But I digress...

Let's not. Let's concentrate on RT's builds and using FC the way it was meant to be.

Closing ticket now, it's all off-topic.
Will reopen if new build still shows same error, but knowing the quality of RT's fixes I am sure it is fixed.

ceremcem, 2021-02-21:
From the user perspective, "downloading an AppImage" seems less scary, I agree. 

Topologically there is zero difference from the point of view of the end user between compiling a software from scratch and dowloading the binary from a server *if the compilation process is straightforward*, like good Docker containers are. Either way you are following 3-4 instructions, that's it. 

Anyway, I was thinking about adding an AppImage creator daemon to the current build environment, which will produce a new AppImage on every commit of LinkStage3 or Assembly3. I had struggled with https://github.com/realthunder/FreeCADMakeImage but I might try it again some time later. The goal is the same. End users will get immediate AppImage updates in 10m to 1h period after a fix.

luzpaz, 2021-02-21:
@ceremcem this would be awesome for master as well....maybe not every commit instead running one at the end of the day?

realthunder, 2021-02-22:
I have created an [issue](realthunder/FreeCADMakeImage#7) showing how I build release images as an example of using FreeCADMakeImage.

snelweg, 2021-02-24:
Using the latest AppImage I can confirm that this is fixed now.

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0224.23964 +3289 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 2dd87c255154b308a7681c1a7c5740151f340b0a
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

ceremcem, 2021-02-24:
@snelweg Why did you reopen then? :)

snelweg, 2021-02-24:
> From the user perspective, "downloading an AppImage" seems less scary, I agree.

Why are you being so smug about this?  You never learned how to assemble your keyboard, but still you are happy to use it. Stop making users into something they do not want to be, we are users and that's it. If you want to compile go ahead but do not expect users to do that. And btw it is a little belligerent to boast about how easy it is, that's not the point.

> Topologically there is zero difference from the point of view of the end user between compiling a software from scratch and dowloading the binary from a server _if the compilation process is straightforward_, like good Docker containers are. Either way you are following 3-4 instructions, that's it.

Wow, you really do not stop do you.

> Anyway, I was thinking about adding an AppImage creator daemon to the current build environment, which will produce a new AppImage on every commit of LinkStage3 or Assembly3. I had struggled with [...]

That would be real nice. Users will be grateful.

I hope this topic is now closed, the fix worked and the latest AppImage works as it should regarding this issue.

closed 2021-02-24 by snelweg

ceremcem, 2021-02-24:
> Why are you being so smug about this?

Sorry, I didn't mean to. I was trying to explain *my* perspective, that was
it. Sorry about that.

luzpaz, 2021-02-24:
snelweg, hope you don't take this as a command but more as a friendly suggestion, please chill out.