realthunder/FreeCAD_assembly3 #466: Missing objects in Wevefront OBJ file if Sub-ShapeBinder is used

Rabbit-sk, 2021-02-16:
Dear RealThunder,

I found out that some objects which are referenced to other bodies by Sub-ShapeBinder are not present in export file. 
Here is project file where it occurs:
[CubeNotExported.zip](https://github.com/realthunder/FreeCAD_assembly3/files/5987412/CubeNotExported.zip)

[image]
[image]

After detaching Sub-ShapeBinders in  Body002, cube is exported.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

realthunder, 2021-02-16:
This is not really a bug. The role of SubShapeBinder inside a body is similar to datum. They are meant to be used by other features. Currently, the body only supports to export the shape of the current tip feature, i.e. the one marked by the green square in the middle of its the tag icon. If you really want to export the shapes from other solid features or binders inside a body, you can create a Part and then add those objects through link, or relative link if you want to maintain their relative placement. If it is just a one time export, you can simply select those features individually and then export.

Rabbit-sk, 2021-02-16:
I think there is a misunderstanding. I create object1, then object2 in which some face/s of object1 is/are referenced by SubShapeBinder/s, because I want to follow up on its shape. Then I select all objects, and some of them are not exported. I know that SubShapeBinders are for using by other features as pad, pocket, sketches... to create another solid shape. I do not want to separately export solid features in the middle of history of body, or SubShapeBinder shapes. I want to export all selected bodies Like this: 
[image]
and their final state represented by they last solid feature marked with Tip. But this is not working in the case of that cube and some other objects, which was used by another bodies through SubShapeBinders. After detaching those SubShapeBinders it works.

BTW that cube has Tip on the right place:
[image]

realthunder, 2021-02-16:
I get what you mean now. That's an unexpected error.

Rabbit-sk, 2021-02-22:
I deleted FreeCAD folder, extracted it again, substitued the `ArchCommands.py` file in  `../Mod/Arch/` and also copied it to `../Mod/Draft/` (despite it seems that it should not be there), as you suggested [here](https://github.com/realthunder/FreeCAD_assembly3/issues/473#issuecomment-782958617) and exporting is still broken :(
[image]
[image]

realthunder, 2021-02-22:
Oh, that's a typo. Sorry about that. It should be in `Mod/Arch/` overwriting the existing one.

Rabbit-sk, 2021-02-22:
I thought. I copied it to the `../Mod/Arch/`, but export still doesn't work fully.

ceremcem, 2021-02-22:
@Rabbit-sk Sure thing. ASAP.

realthunder, 2021-02-22:
Ah, right, that specific change only works with some other early changes. Please search for `Part::SubShapeBinder` inside that `ArchCommands.py` file, and change it to `PartDesign::SubShapeBinder`.

I have moved SubShapeBinder implementation into `Part` module in some early commit.

Rabbit-sk, 2021-02-22:
This change helped. Objects that have not been exported are now being exported, except for a small number of objects. I will check the export for missig objects and try to find out where the problem is. In the meantime, I left this problem open.

Rabbit-sk, 2021-02-22:
Now I used build that I compiled from newest source code, and number of missing objects is smaller. I will look at the remaining non-exported objects and provide more information

Rabbit-sk, 2021-02-22:
There were missing only two objects, which were elso mentioned in ReportView (before latest obj export fix, there were missing also objects that were not mentioned in the ReportView). 

- First object consists of binder of a sketch and Extrusion feature from it. Tip was not set to Extrusion feature as described in the ToolTip. Easy fix was to set Tip to that Extrusion. Why Tip is not set to Extrusion feature? BTW Extrusion feture is extremely usefull when one want to make only a surface.
- Second object consist only of a base feature made from solid shape of another body. I guess that since this BaseFeature is basically a SubShapeBinder, and they are excluded from processing while exporting, such body is not exported. This is fine, there is the Link function to make copies. This approach with BaseFeature without adding further features to it,  was choosen in a hurry... 

I think that this issue could be closed. There is only a question why not to set a Tip to Extrusion feature.

Rabbit-sk, 2021-02-22:
> Sure thing. ASAP.

@ceremcem thank you for your willingness.

closed 2021-02-23 by Rabbit-sk