realthunder/FreeCAD_assembly3 #467: DatumPlane is created despite canceling its creation

Rabbit-sk, 2021-02-18:
Dear RealThunder,

after attempt to create a new `DatumPlane` after unintentionaly selecting a face, which is out of scope, this message shows up:
[image]

`DatumPlane` with `Support` reference, which is out of scope, is created despite canceling the `DatumPlane` creation:

https://user-images.githubusercontent.com/74265320/108410669-000c5a80-7228-11eb-9688-bc1b0bdc4219.mp4

`ReportView` message from the video above:
```
20:25:01  <App> GeoFeatureGroupExtension.cpp(581): Unnamed#DatumPlane.Support(links to Unnamed#Box) is out of scope:Unnamed#Body vs. Unnamed#Body001`
```

I think that a new `DatumPlane` should be not created if its creation dialog is canceled.

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git)
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.6.8
Qt version: 5.12.6
Coin version: 4.0.0a
OCC version: 7.4.0.beta
Locale: Slovak/Slovakia (sk_SK)

Regards

snelweg, 2021-02-22:
Happens here all the time too. Happens with other dialog boxes too I noticed. I will start making notes when this happens.

> OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Rabbit-sk, 2021-02-23:
It works perfectly now.
Verified in the version compiled from the source code after commit [9deefac](https://github.com/realthunder/FreeCAD/commit/9deefac4f2748d4af44d5d856f605c92d736c8c3).

closed 2021-02-23 by Rabbit-sk