realthunder/FreeCAD_assembly3 #470: Crash while hovering on element

ceremcem, 2021-02-19:
# Description 

FreeCAD crashes while hovering on an Element while another element is selected. 

# Screencast

https://user-images.githubusercontent.com/6639874/108476681-63879e00-72a3-11eb-9231-3b6f42e8cf93.mp4

# Traceback
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f285d074730]
#1  0x7f285d554655 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x25
#2  0x7f285d6de108 in SoChildList::traverseInPath(SoAction*, int, int const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x108
#3  0x7f285f670629 in Gui::SoFCSwitch::doAction(SoAction*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0xa19
#4  0x7f285d73f06e in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x48e
#5  0x7f285d73f06e in SoSeparator::GLRenderInPath(SoGLRenderAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x48e
#6  0x7f285d554694 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x64
#7  0x7f285d55b995 in SoGLRenderAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x185
#8  0x7f285d555321 in SoAction::apply(SoPathList const&, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x581
#9  0x7f285f67c5c8 in Gui::SoBoxSelectionRenderActionP::apply(Gui::SoBoxSelectionRenderAction*, SoPathList const&, int) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0xf8
#10  0x7f285f67c6b7 in Gui::SoBoxSelectionRenderAction::apply(SoPathList const&, int) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0xa7
#11  0x7f285d55b224 in SoGLRenderActionP::renderSingle(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x264
#12  0x7f285d55b6b8 in SoGLRenderActionP::render(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x108
#13  0x7f285d55b8af in SoGLRenderAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x9f
#14  0x7f285d554c22 in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x372
#15  0x7f285f67c999 in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x19
#16  0x7f285d7ff7f0 in SoRenderManager::renderScene(SoGLRenderAction*, SoNode*, unsigned int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xd0
#17  0x7f285d800838 in SoRenderManager::actuallyRender(SoGLRenderAction*, int, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x68
#18  0x7f285d8009d2 in SoRenderManager::renderSingle(SoGLRenderAction*, int, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xc2
#19  0x7f285d801045 in SoRenderManager::render(SoGLRenderAction*, int, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x105
#20  0x7f285d7fefcb in SoRenderManager::render(int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x8b
#21  0x7f285f732151 in Gui::View3DInventorViewer::renderScene() from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x231
#22  0x7f285f6efced in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x2ed
#23  0x7f285f6f7032 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x32
#24  0x7f285c8264d8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d8
#25  0x7f285c8c9cde in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#26  0x7f285cb12d4b in QGraphicsView::viewportEvent(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x16b
#27  0x7f285f6efda7 in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x87
#28  0x7f285be6d19b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#29  0x7f285c7e84b1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x71
#30  0x7f285c7ef970 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#31  0x7f285f406ca8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x68
#32  0x7f285be6d489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#33  0x7f285c81f0ca in QWidgetPrivate::sendPaintEvent(QRegion const&) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3a
#34  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x16e5b2) [0x7f285c7f75b2]
#35  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x16ee19) [0x7f285c7f7e19]
#36  0x7f285c80e9d5 in QWidgetPrivate::syncBackingStore() from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa5
#37  0x7f285c826608 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x308
#38  0x7f285c92cdd4 in QMainWindow::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x64
#39  0x7f285f7eca46 in Gui::MainWindow::event(QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x66
#40  0x7f285c7e84c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
#41  0x7f285c7ef970 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#42  0x7f285f406ca8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x68
#43  0x7f285be6d489 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#44  0x7f285be7046b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x1cb
#45  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2bc103) [0x7f285bebf103]
#46  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2ae) [0x7f285892df2e]
#47  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4e1c8) [0x7f285892e1c8]
#48  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f285892e25c]
#49  0x7f285bebe727 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x67
#50  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0xdd491) [0x7f28520c4491]
#51  0x7f285be6c15b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#52  0x7f285be74132 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#53  0x7f285f377a1e in Gui::Application::runApplication() from /home/aea/fc-build/Release/lib/libFreeCADGui.so+0x31e
#54  /home/aea/fc-build/Release/bin/FreeCAD(+0x4850) [0x55d13ed36850]
#55  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f285b4d809b]
#56  /home/aea/fc-build/Release/bin/FreeCAD(+0x4b4a) [0x55d13ed36b4a]
```

## Version 

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.26865 (Git)
Build type: Release
Branch: LinkStage3
Hash: f7007966e1f73a423bef124c114587768f655609
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

The same model I've sent to your Gmail, subject was "...related to 460"

realthunder, 2021-02-19:
This problem has been fixed by realthunder/FreeCAD@93e68ca125d

ceremcem, 2021-02-19:
Ah, okay then. I had rolled back because of #460, so I missed it.

closed 2021-02-19 by ceremcem