realthunder/FreeCAD_assembly3 #471: Space Mouse support in DMGs

mgrindel, 2021-02-20:
Firstly I have to express my gratitude for sharing your extensive work with all!
Although I like all the functionality you have added I do not use your fork due to the lack of Space Mouse support...
I know I can compile but would you please consider including support in your future pre-compiled releases?
Many thanks for your consideration

realthunder, 2021-02-20:
I am also puzzled as to why many complain about trouble using space mouse using my branch. I don't recall any specific changes about this. Anyway, I've just purchased one from Amazon. Hopefully I can solve this problem once and for all.

mgrindel, 2021-02-20:
Wow! I was not expecting a response so soon! Your response is much appreciated :-)
From what I've read you simply need to have the drivers installed when you compile and support will be there.

luzpaz, 2021-02-21:
@mgrindel do you have issues with spacemouse on FreeCAD 0.19_pre master?

mgrindel, 2021-02-21:
> @mgrindel do you have issues with spacemouse on FreeCAD 0.19_pre master?

Hi, No problems at all with the last few versions on FreeCAD 0.19_pre master... :-)

snelweg, 2021-02-22:
I still have sometimes problems that my spacanavigator stops working in Debian AppImages. It happens in all versions since I switched over to LinkStage3 branch. There is no clear message or situation where this occurs, it looks random.
Sometimes I can get it to work again after restarting the spacenavd deamon, but mostly I have to restart FC to get a functional navigator.
Sorry to not being able to be more specific, as I said there is no clear moment this happens.

mgrindel, 2021-02-22:
Aha yes I have found this on rare occasions in OSX as well.
Usual only with very big modules and while moving the mouse and the space mouse simultaneously....
Mostly pausing/waiting for a short while will resume function but yes sometimes I have to restart fc to fix.

PS today's build is really smooth! Love it....

⁣Sent from Blue ​

realthunder, 2021-03-07:
> @luzpaz Hi, No problems at all with the last few versions on FreeCAD 0.19_pre master... :-)

My space mouse has arrived, and strangely it does not work with upstream FreeCAD 0.19 MacOSX build. I found a workaround referenced above to make it work on my test machine.

mgrindel, 2021-03-07:
Hmm that's odd?You did load the driver? :-)The last 4 releases have been fine..... using FreeCAD_0.19-24267-macOS-x86_64-conda.dmg at the moment...Best regardsMark

realthunder, 2021-03-08:
> Hmm that's odd?You did load the driver? :-)The last 4 releases have been fine

Hmm... Maybe it's because 3DConnexion has recently upgraded their driver which causes it to break? The linked driver download address in upstream conda recipe has expired. I manually download and install the latest driver. Tested with upstream FreeCAD image and it does not work.

luzpaz, 2021-08-04:
JFYI, there's an upstream PR related to spacemouse https://github.com/FreeCAD/FreeCAD/pull/4924